UCSBarchlab/PyRTL #107: Reworking Muxes in Pyrtl

Gamrix, 2015-10-03:
So Tim @timsherwood  and I were discussing the mux currently in helperfuncs. Currently, the mux has this function definition:

```python
def mux(select, falsecase, truecase, *rest, **kwargs):
    """ Multiplexer returning falsecase for select==0, otherwise truecase.

    :param WireVector select: used as the select input to the multiplexor
    :param WireVector falsecase: the wirevector selected if select==0
    :param WireVector truecase: the wirevector selected if select==1
    :param additional WireVector arguments *rest: wirevectors selected when select>1
    :param additional WireVector arguments **default: keyword arg "default"
    :return: WireVector of length of the longest input (not including select)”””
```

For muxes of size greater than 2 I believe that this function definition is way too complex and intimidating to beginners. I believe that it should be changed to:

```python
def mux(select, *ins, **kwargs):
    """ Multiplexer

    :param WireVector select: used as the select input to the multiplexor
    :param WireVector arguments *ins
    :param additional WireVector arguments **default: keyword arg "default"
    :return: WireVector of length of the longest input (not including select)”””
```

And for people who want labeled true and false cases, a separate function (currently proposed to be Select but we are still looking for a better name) could be made.

```python
def select(select, truecase, falsecase):
    """ Multiplexer returning falsecase for select==0, otherwise truecase.

    :param WireVector select: used as the select input to the multiplexor
    :param WireVector truecase: the wirevector selected if select==1
    :param WireVector falsecase: the wirevector selected if select==0”””
```

Arguments for this:
                Makes the function much easier for a beginner to interpret
                Clears up the confusion when it comes to what truecase means in muxes of size greater than 2

Arguments against:
                We are making two different functions for very similar functionality (ref: Zen of Python)

Backwards Compatibility:
                With this, I would propose to allow falsecase and truecase to be accepted as kwargs for the near term (and to still work correctly in that case), but also print out a Deprecation Warning, noting that these will be phased out in the next pyrtl release

Alternatives:
Keep it as is
Do not implement the select option
                (allays the problem of having multiple identical function)
                Tim thinks that beginners from CE backgrounds would often get muxes wrong

timsherwood, 2015-10-03:
This is a really nice description of the choices.  The only other reason I would put in the "arguments against" is the extensive use of "mux" in typical introduction to computer engineering books rather than "select" which would be a new thing.  However, after sleeping on it I think this is probably the right way to go if we can find a good name.  

The name "select" conflicts with the select operator which breaks out a slice of wires from a wirevector.  Other options would be "choose", "pick", or "conditional".  I would lean towards "conditional" or "cond" since the ternary operator (which this function closely resembles) is also called the "conditional operator".

timsherwood, 2015-12-23:
What do people think: "select", "ternop", "predmux", "hw_if", ....

Gamrix, 2015-12-23:
I really like "select" and I also wouldn't mind if it is called "predmux". "ternop" seems way too vague, almost like a name for a ternary operator wrapper. "hw_if" would be more suitable as a alias for ConditionalAssignment than mux, as it sounds like it would make more sense in that context.

I would also be fine with the name "mux_2" or a variant of that.

~~Looking back at the proposal, I would also be fine with this function being just a single function if we change around the  function definition to make it more readable, eg:~~

```python
def mux(select, *ins, **kwargs):
````
~~where you can either specify 'truecase' and 'falsecase' or pass in wires for the *ins but not both. This option would simplify the description of the mux without separating the operation into two separate ops.~~

Actually, after typing up the strikeout portion, I changed my mind. Combining two different but related functions into one function just for the sake of combining them is a bit silly, especially if we will have two different descriptions of how to use the same function. However, it is still better than the monstrosity that is the current mux function.

jemcmahan13, 2015-12-24:
`select` is good, and is rooted in the idea of a HW selector. Unfortunately Oct 3rd Tim says it conflicts with the "select" keyword. `ternop` and `predmux` seem a little confusing; why should a "predumx" be different from "mux"?

It's been a while since we've talked about this, so maybe this was already addressed: why not have one function where "falsecase" and "truecase" are part of the **kwargs? If there's a separate function where the keywords are mandatory, it's still a setup where their usage is optional, since you choose to use the `select` version over the `mux` version. Why not use a single function where they're optional?

Gamrix, 2015-12-25:
The whole idea of this refactor is to is to make it simpler for users to understand what these functions are doing, and having one function take in two almost orthogonal sets of arguments depending on what you want to do with it could be potentially confusing. More importantly though, we will always have a mux_2 variant of the function, as the generalized mux uses it in multiple places (as it does in the current code). Therefore, it seems silly to have one public function that is complicated to understand, complex, and messy under the hood as well as one private function when we could just have two public functions that are simple. 

Kwargs, while making the function definition simpler, would make the task making the mux both be correct as well as cover the edge cases (eg only supplying one of falsecase or truecase) harder than it really should be.

timsherwood, 2015-12-28:
Okay, let's go with `select`.  We can always change the underlying representation if that is confusing with a refactor.  I agree that it is probably most important that the user facing part is clear.

Gamrix, 2016-01-08:
fixed in commit number 6862f03d1

closed 2016-01-08 by Gamrix