UCSBarchlab/PyRTL #116: Multiplication Op not synthesizable

Gamrix, 2015-12-10:
I believe that sooner or later, we should implement synthesis for the multiplication op. One question would be which multiplication operator would we use by default @timsherwood .

Gamrix, 2015-12-13:
The main way to do this as I and @timsherwood have discussed is by having a presynthesis pass to reduce more complex hardware (like '+' and '*'). This shouldn't be hard once I move the transforms that are in PyRTL research into PyRTL core. Transforms will be very useful for this task and it will also be a good real world usage of that code. If anyone wants to take this on, please let me know, and I'll get code moved around to facilitate fixing this.

abe-k, 2016-04-22:
It looks like synthesis already handles `+`, `<`, etc., and that `*` is the only operator still unsynthesizable.

The `_basic_mult` provided uses `+`, so either it would have to be rewritten to only use simple single-bit operations, or it would have to be applied before `_decompose`.

Does anyone see a problem with (at the beginning of `synthesize`) copying the input block and using `net_transform` to replace the `*` operators with `_basic_mult`, and then using this modified block for the remaining synthesis?

Gamrix, 2016-04-22:
Nope, I don't see any problems with that.

Part of what we (ok, really I) want is to not have implementations of '+-<>=' and possibly 'x' that can only be used for synthesis. The current implementations of those ops in synthesis are rather ugly and unreadable as well. Therefore, after you implement the synthesis of `_basic_mult` with `net_transform`, I would like it if you could write normal PyRTL implementations of '-<>='  and use them as well as existing implementations of '+' and mux in place of the current synthesis.

abe-k, 2016-04-24:
I ran into a very strange set of issues when using `copy_block`, and I think I've finally isolated it to one: the `copy_block` code uses `Block.__iter__`, which will throw an error if there is a loop, so the tests for loop detection fail early. If I remove the check in `Block.__iter__`, I somehow get a lot of `Wires declared but not connected` from the sanity check when setting the working block in the loop detection tests.

Is it a problem if loops are detected during synthesis (i.e. that the test cases for loop detection are rewritten)? Otherwise, how to go about copying a block?

Gamrix, 2016-04-25:
I would recommend changing `for net in block_in:` to `for net in block_in.logic:` instead. Block.__iter__ relies on the block being an acyclic graph (aside from registers), and there is no reason why we need to keep that constraint for copying an arbitrary block. Also, you should see a performance improvement as there are no longer extra calculations to create an ordering for the blocks. 

I am interested to see what was the code you used to generate the problematic block.

abe-k, 2016-04-25:
The change of adding `.logic` fixed that problem, thanks. I didn't create the problematic blocks—they're the ones from `TestLoopDetection`.

Now, pylint is complaining about `io_map` not always being present on a Block, but since `copy_block` adds one it will be there when needed. Would it be reasonable to have `Block.__init__` create an empty `io_map` by default to silence pylint?

Gamrix, 2016-04-25:
Yeah, I never thought to try to test those. I guess they are still technically valid blocks (though they are not simulatable). I think that there might need to be a small discussion on whether we should specify whether certain functions should be usable on non-simulatable blocks (as there is a reasonable argument as to why one would want to make such a block.

For the `io_map` question, I don't know. It might make sense as a child class of Block ( from which PostSynthBlock can inherit from), though I would hesitate to recommend adding more classes into the block hierarchy. Making a decision like that would be something more suited for next week's meeting. Therefore, feel free to just silence the warning on a single line basis (pylint comment) until then.

abe-k, 2016-04-25:
Multiplication synthesizes and simulates correctly! Now to write tests.

Where should I put these tests, and is there anything in particular I should pay attention to when writing them?

Gamrix, 2016-04-25:
I would recommend putting some code into the test_passes, as well as some simulation code into test_simulation

things that might be interesting are: check if x * y * z synthesizes and simulates properly
Make sure that stuff properly simulates.
maybe figure out the number of logicnets a smaller multiplier design should have, and verify that there are that many multipliers

abe-k, 2016-04-25:
I have some tests written for synthesis of multiplication. There already was a test for simulation, so I left that as is. I don't think it make sense to test the number of logic nets, as that's more of an implementation detail rather than an interface.

One annoying thing is that `_basic_mult` (and the other operator implementations) assume they're working in the working block, which currently isn't the case when called from `synthesize`. I worked around it by passing the block in to `_basic_mult`, but would it make more sense to temporarily change the working block at the beginning of `synthesize`?

Gamrix, 2016-04-25:
I would say that it would make more sense to temporarily change the working block. A lot of arithmetic code assumes the working block, so assigning a temporary one would avoid a lot of hand wringing.

abe-k, 2016-04-25:
I've pushed the multiplication synthesis to my fork: https://github.com/abe-k/PyRTL/commit/758d0ec0c7be48c236df613689d5828ea3f53da1

Before doing a pull request, I think I'll try adding clean support for `+-<>=x` as you suggested. There's a `mux` in `helperfuncs.py`, but which adder should I use? (Also, the adders are in `rtllib`, which I assume we don't want the core to depend upon, so the question is really which adder I should copy over.)

Gamrix, 2016-04-25:
Just copy the old ripple carry adder from example_2 into helperfunctions.

abe-k, 2016-04-25:
That's all the advanced operators implemented in standard PyRTL and used in `synthesize`: https://github.com/abe-k/PyRTL/commit/8182a8f0d6041349b2d66a411f8f6a3e689dbe27

I think some tests still need to be written for synthesis of `'x'` and the comparison operators, but it's almost there.

Gamrix, 2016-04-26:
Sounds good. Also, feel free to delete the code you commented out. We can always look back in history to get it if we need it.

abe-k, 2016-04-26:
Good thing I added the tests, since I'd gotten the implementation of `'x'` reversed. Pull request has been created: https://github.com/UCSBarchlab/PyRTL/pull/134

closed 2016-04-27 by Gamrix