UCSBarchlab/PyRTL #118: Some Verilog-style constants not accepted in Const

jemcmahan13, 2015-12-17:
Const will accept Verilog style strings only if they specify no base or use binary or octal bases ('b' or 'o'). Bases 'h' and 'd' are all also valid in Verilog, but the function `_convert_verilog_str` relies on the Python built-in number base conversion, which supports 'b' and 'o', but not 'h' or 'd'.

```python
x = Const("5'b10011")  # accepted
x = Const("32'd12")  # rejected
x = Const("10'o7")  # accepted
x = Const("8'h7F")  # rejected
```

abe-k, 2016-04-20:
It looks like Verilog supports some other features in its integer literals as well (according to http://web.engr.oregonstate.edu/~traylor/ece474/lecture_verilog/beamer/verilog_number_literals.pdf). Do we want to support any of signedness, underscore-separators, or default values?

timsherwood, 2016-04-20:
It would be fine if signed is supported, but it should throw an error deeper in the const code (signed values is a discussion for another day).  It is also fine if we throw an error here for now.

Underscores are awesome and it would be great if they were supported

As to default widths I think we would be best to avoid supporting them -- explicit is probably better here and the idea that 32-bits is the default default is perhaps no longer valid (not sure if Verilog would make that same choice if they had to do it over).

abe-k, 2016-04-20:
I think I'll add support for negative literals (two's complement), but an error if the S prefix for signedness is used.

timsherwood, 2016-04-20:
Just be aware that there is some very unintuitive behavior if we add signed constants as twos complement without proper sign types:   e.g. Const('-1') > 0 is true; Const(-1).extended(32) will not be what you think; etc.

It probably merits some discussion while John is still around. :)

Perhaps it make sense to add a special "twos_complement" function or something like that in the mean time?  I agree that it is a PITB to have to do the two's complement calculation yourself though.

abe-k, 2016-04-20:
We already support negative integers being passed, as in `Const(-1, bitwidth=32)`. This just extends that support to Verilog-style strings, as in `Const("-32'd1")`.

timsherwood, 2016-04-20:
Sound good then --  thanks for clarifying.

timsherwood, 2016-05-01:
Is this one done?  Can we close this issue?

closed 2016-05-01 by Gamrix