UCSBarchlab/PyRTL #119: Simulation and FastSimulation interface not the same

jemcmahan13, 2015-12-19:
`FastSimulation` uses the attribute `context` for accessing the map of {signal -> value}, but `Simulation` does not. It looks like it uses `value` instead. I haven't tested other things, but the interfaces should be checked and changed to be as equal as possible.

timsherwood, 2015-12-19:
Context and value are different (not just in name) for implementation reasons: .context is a map from string to simulation value, while .value is map from wirevector to simulation value.  You are reaching into the implementation there.  Do you just want to read the value or do you need to modify it?  I was going to add a unified interface, but then I realized that you should probably read it from "trace" rather than the simulation directly.  If I add an interface to trace would that help, or do you think it needs to be on the simulation itself?

jemcmahan13, 2015-12-20:
I've used this in a few PyRTL projects, but it's only ever for one reason: some wrapper script reads in a program, loads it into a memory, and runs the hardware while checking and printing output values each cycle. It also needs to be able to read internal signals for debugging (a wave trace won't show you stack contents). So no, I don't ever modify any values.

I didn't realize that `trace` had a similar capability. It doesn't make a difference to me which object has the interface; the only use I can see is to have a read-only map each cycle from wire (or name) to value. Being able to switch `Simulation` and `FastSimulation` without having to modify other parts of the program would be really nice: `FastSimulation` is faster for real (long) programs, but `Simulation` typically runs small test programs faster without the long initial JIT.

Being able to mutate the values is interesting, but we've already put some effort into creating transforms that do things like insert random errors, so the use cases might be covered elsewhere.

timsherwood, 2015-12-21:
If you want to look through the memory state, then the trace is probably not the right place to do that.  I added "inspect' as a method to both simulation and fastsim, which should unify the interfaces.  Instead of ```python sim.value[foo]``` it is now ```python sim.inspect(foo)```.  It should work in both and test now cover it.

commit f8a17b17 (2015-12-21): adds inspect to sim and fastsim, fixes #119
closed 2015-12-21 by timsherwood