UCSBarchlab/PyRTL #120: Mux error on `with` predicate having incorrect bitwidth

jemcmahan13, 2015-12-21:
```python
from pyrtl import *

x, y = WireVector(8), WireVector(8)

with conditional_assignment:
    with x:
        y |= Const(5)
```

Might be confusing to a new user who doesn't expect a mux error when they have no (explicit) muxes.

timsherwood, 2015-12-21:
This is actually pointing to a huge and devious bug in PyRTL!  In some instances a predicate (e.g. ```x``` in ```with x```) will be automatically transformed to 1-bit wire, while in other cases it is not.   In fact even the unit test suite is not-consistent it it's assumptions about how to treat these predicates!   I ran into errors in the unit test trying to fix it both ways ("casting" to a 1-bit wire and throwing an error).  Crazy.  So the question is what is the "right" thing to do for PyRTL.

Option 1: ```with x``` throws a clear error when len(x) is not 1.
Option 2:```with x``` takes an "or" of the bits of (x) when length > 1 (C and Verilog style).

```python
with ~x:
    foo |= bar
```

is fine.  But 

```python
with x:
    foo |= bar
```
becomes

```python
with or_all_bits(x):
    foo |= bar
```

Which is more explicit, but decreases readability (and introduces the possibility of someone saying "and_all_bits" by mistake.

Gamrix, 2015-12-21:
After thinking about it for a long time, I am more in favor with the second option. Not only does it act like a value would in other languages (not only C and Verilog, but also for example python itself), but more importantly, having to look for or make a "or_all_bits" function would be a big challenge for a beginner to PyRTL, and would increase the amount of knowledge necessary to create things in PyRTL.

jemcmahan13, 2015-12-21:
I think the fewer weird implicit behaviors we can have baked in, the better. Something akin to python's `any` and `all` is exactly what we're talking about; there, explicit is better. I don't think there's any challenge to having helper functions that do unary reduction (which Verilog does have).

On the note of have to "look for" functions, I find myself looking up function alot --- because I can't remember names or ordering, or can't remember if the thing I'm thinking of exists or not. I just grep source; how is looking up functions supposed to work?

Gamrix, 2015-12-21:
I guess the real question is how often we expect people to do something like this. If it is often (something around 10% of the time), I think it is worth it to have implicit behavior. However if this rarely occurs in real code, then I would think that maybe it is better to throw an error.

Also, I forgot that coders need not find the or_all_bits function. x != 0 does the same thing (though arguably less efficiently unless thrown through optimization).

I rarely run searches through codebases. Usually, when I am not familiar with a codebase, I usually don't feel comfortable enough with grepping through the source, and instead resort to browsing the docs.

timsherwood, 2015-12-21:
Actually, this conversation made me realize that there is a fatal error with option 1.  In fact my example 

```python
with ~x:
    foo |= bar
```

is totally wrong!  If len(x) is 2 and x==1, then both `x` and `~x` will evaluate to true because ~ is bitwise not.  Not cool.  Option 2 must be the winner to prevent this sort of bug in the future, although John is absolutely right that `x==0` and `x!=0` are way better than `or_all_bits` and should be the preferred method of reducing to a boolean.

commit 1f5bac69 (2015-12-22): condition only on wirevector of len 1, fixes #120
closed 2015-12-22 by timsherwood