UCSBarchlab/PyRTL #123: Move timing.py to rtllib???

Gamrix, 2016-01-04:
I feel as though timing.py is not essential or integral enough to be part of the core library. Yes, there will be people who will need it, but people will often need a lot of different things in rtllib. Also, as there are no core dependencies on timing, it could be easily moved to rtllib.

Namespace are a good honking idea, and I feel by filling up the core namespace too much, people will not notice the timing functions. We should give them their own namespace so that people can find a central location for all of the timing functions, and a great place for that is rtllib, where each file is its own namespace (Core has departed namespace internally, but it exposes itself to users as a single namespace).

timsherwood, 2016-01-04:
I still can't get rtllib to execute for me outside of tox?  /pyrtl is in my pythonpath, do I need to add /pyrtl/rtlib explicitly?

Gamrix, 2016-01-04:
no, you should only need to do something like ```from pyrtl.rtllib import adders```

Gamrix, 2016-01-11:
Moved to the analysis folder in commit ea99ee811a59158e723869ebdd6b18456def81ec

closed 2016-01-11 by Gamrix