UCSBarchlab/PyRTL #125: Fastsim failure for item with a name with a space in it

Gamrix, 2016-01-25:
If you name a item with a space in it, then fastsimuation fails to compile it as spaces cause the generated code to have a syntax error.

Proposed fixes:
Change the varname to replace spaces with underscores
* Simple but will slow down fastsim
Throw an error when the we find a name with a space in it
* doesn't impact performance as much (still a performance hit)
* further restricts the names people can give to items.

Any fix that happens needs to be accompanied with additional unittests.

timsherwood, 2016-01-25:
The problem is more general than spaces.  Right now PyRTL allows any number of illegal things as variable names including brackets, escapes, etc.  Hmm..  maybe we should use a hash of the names?

Gamrix, 2016-01-25:
Yeah, there are a lot of symbols that aren't part of a legal identifiers. However, I wouldn't expect most of them to be used in a name. But that is not enough of a reason to ignore the problem entirely.

One thing that we do have to consider is that as long as part of fastsim is not converted into bytecode, making the fastsim code for generating and validating names slower will result in a large performance hit.

Gamrix, 2016-06-08:
Fixed in 5a2f433 (look at issue #148)

closed 2016-06-08 by Gamrix