UCSBarchlab/PyRTL #127: MemIndexed has no name

jemcmahan13, 2016-02-02:
`MemIndexed` inherits from `WireVector`, but `super` is never called and no name is assigned, so attempting to access its name results in an error. This may have been intentional because it refers to a "port" and not a "wire", but it can cause confusing behavior, as a user shouldn't be aware that they're handling a `MemIndexed` instead of a `WireVector`.

```python
mem = MemBlock(8, 8)
x = mem[Const(2)]
y = x + 1
print y.name  # this is fine
print x.name  # this is an error
```

timsherwood, 2016-02-02:
Interesting, this has implications for printing and debug as well.  We can't just call __init__ because it will register it as a wire with the block.

Gamrix, 2016-07-04:
On the other hand, now that we can find wires in simulation based on their name, is it really smart to expose a name for something that cannot actually be accessed? I feel like having no name might actually be better in some cases. 

(another option is to give it a name such as `"Generic Mem Read Port"` though this too might not be enough to deter people from trying to use it in simulations)

jemcmahan13, 2016-07-04:
It's not even necessarily a Read Port though --- the reason to have the `MemIndexed` object is because it's not yet clear if the memory access will used for a read or a write.

Having no name for it makes sense and solves some problems --- it's not actually a wire, so why treat it like one --- but then exposes some weird behaviors (like the original except). If we treat it differently from a wire, we have to be prepared to expose that difference to the user, meaning for certain behavior you have to write kind of weird code:
```python
x = mem[addr]
y = WireVector(width)
y <<= x
print y.name
```

Because of that, I'm in favor of it transparently acting like a wire. Why *not* allow it to be used in simulation? 

If names are the only place where trouble is arising for `MemIndexed`, then another solution is just to have custom code that deals with names on memory ports but prevents it from appearing in the simulation.

Gamrix, 2016-07-13:
in #171 we changed `_MemIndexed` so that it now gets and sets the name of the underlying wire. If there is no underlying wire yet, it will build one.

closed 2016-07-13 by Gamrix