UCSBarchlab/PyRTL #129: Conditional Assignment section in docs?

jemcmahan13, 2016-02-11:
I can't find the conditional_update interface in the docs. It looks like the entire section is missing. I'm searching for a function to get the current predicate inside of a `with` block.

Gamrix, 2016-02-17:
Why exactly do you need to do this???

Also, ask Sherwood what is happening in those bowels. Conditional Assignment is the only part of the codebase that I don't dare touch.

jemcmahan13, 2016-02-17:
The issue exists to document a lack of something important in the docs.

Getting the current predicate is useful if your building something that sits on top of pyrtl and/or interacts with it. E.g., you want to save an ordered list of predicates to enforce precedence across `with` blocks. or you want to modify the predicates or do additional operations under that condition.

commit fcf1c324 (2017-05-13): adds docs for conditional assignment, fixes #129
closed 2017-05-13 by timsherwood