UCSBarchlab/PyRTL #137: Input and Output shared between blocks

abe-k, 2016-04-27:
To avoid needing `io_map` and the contortions involved in mapping interfaces between pre- and post-synthesis blocks, we agreed in the meeting to make Input and Output WireVectors be shareable between blocks, so that pre- and post-synthesis blocks could have the same connections.

The major problem is that currently, WireVectors store a reference to the block they belong to, and use that reference when creating LogicNets. Rather than specializing just Input and Output to (somehow) not need to reference a specific block, would it be reasonable to make all WireVectors simply use the working block for construction? (Still having the sanity check only allow I/O to be shared.) If not, how should existing in multiple blocks be handled?

timsherwood, 2016-04-28:
Hmmm..  this is a problem. I don't see a good way forward honestly.  I don't think we want wirevectors to only use the working block for construction since it will make working with multiple block really hard.  Eliminating the .block attribute from just Input and Output might work but this really goes against be inheritance practices (of extending functionality rather than removing it).  Maybe we could make .block a property which Input and Output throw an error if .block is required? but that in turn will probably restrict the set of things you can do on inputs and outputs. 

I think we might need to go back to the drawing board to figure something else out.

abe-k, 2016-04-28:
What if (and this might be a substantial refactor) we say that only Input and Output don't have a block, and any operation that's done will figure out the block from the other WireVectors involved, which will still have associated blocks. In the edge case that the operation only involves Inputs and Outputs, the working block will be used. We can get around inheritance issues by factoring out most of WireVector into a common base class, and have "has a `.block`" be the defining feature of the `WireVector` subclass.

Gamrix, 2016-04-28:
One thing to note is that I have refactored `set_working_block` earlier today so that it can be used with the `with` statement (eg `with set_working_block(block_obj): ... `)

I don't think that it would be too unintuitive to force people to only build on the working block. A person should to separate different blocks anyways as creation of wires from scratch is usually cleanest when you are doing it with the working block. Besides, it would be bad if a person was building nets for a different block than they anticipated. I cold imagine myself accidently creating a circuit in the wrong block because I didn't know that new blocks and wires are made based on the block of the previous wires, and not based on the working block.

Another argument (maybe it is the same argument) is that the working block is explicit, whereas using the block built into the wires could not be if there are multiple blocks involved.

The one downside is that by removing `.block` you make it so that a reference to the block is necessary to add more logic to the block. However, I don't see the big harm in making that a requirement.

timsherwood, 2016-04-28:
The set_working_block context seems lie a nice idea (assuming it puts it back like it found it when done). 

I think both removing the .block and/or changing the inheritance structure of such a critical and ubiquitous set of objects makes me very nervous and it is hard for me to justify such a significant change to support the interface.  Are there any other options for supporting the iomapping of these blocks that does not violate the property of blocks and wirevectors?  Can we define an interface (which is just a collections of input and output wires) and operations such as simulation operate on the interface?

abe-k, 2016-04-28:
Having an Interface class seems like a good idea, but I don't see that it solves anything.

Is the idea that for Input and Output, the Interface they belong to would be their `.block`? That doesn't seem very elegant, but otherwise, we still have the same problems of either inheritance changes or ambiguity when doing operations on I/O wirevectors.

The cleanest solution mentioned so far still seems to be removing `.block` and using the working block. Yes, it would be a major change, but would it really be a bad one? After all, if you're defining logic, you're probably creating Consts, Registers, etc. at the same time, so you need to either be in the working block or be passing the effectively-the-working-block around everywhere.

timsherwood, 2016-04-28:
No I am suggesting that we don't share Input Output wires between blocks at all and instead, going back to the original problem we had, see if we can find another way around.

As to the second point, it is a major change which may or may not break a bunch of code (we would have to see -- in the same way our last idea had unintended consequences) which I am sure either way we could fix but in the end, is the system really better for it?  That is what I want to understand.  Right now working_block works as a "convenient default" parameter for the creation of hardware.  Changing it to a "required set of state" changes more than just the lines of code it touches, it gives up something philosophically (moving explicit arguments to be more implicit).  Such a shift to save a couple dictionary lookups later when we want to simulate is not a clear design win to me yet -- especially when I am not sure we have exhausted other options.

Gamrix, 2016-04-28:
The problem is that working_block is not really a convenient default. Instead, it is one of two competing systems deciding what is the block that is used to create the new hardware. Some of the hardware is created based on the `.block` (look at all the places that use the `get_block` method or check `.block` directly). Other pieces of hardware uses the working block (eg WireVectors). I feel that having the choice of blocks split between the two systems does not help with peoples' intuition of and the consistency of the system. Instead, I think it will cause people to get confused when errors occur not at the place closest to where the initial error occurs, but instead much later, where the conflict between the two systems finally collide.

TLDR: I don't think that the current system is actually good. Breaking it might involve some pain, but it will prevent errors in the future.

timsherwood, 2016-05-01:
I am open to this, but I want to understand what you are thinking John with respect to how blocks will be managed by the user.  What does using working block as a context do on __exit__ (is there a stack or is it a just flat)?  Will the expectation be that working blocks are always managed with contexts or that you mix and match?

Gamrix, 2016-05-01:
On exit, the block that was the working block before the context manager is restored as the working block. It is basically a stack as each `set_working_block` stores its own original working block. (look at the test case that I recently committed) The expectation will be that the user uses the context manager when temporarily changing working blocks, and uses the `function` directly whenever they want to switch to another working block 'permanently', (eg for synthesis).

abe-k, 2016-05-04:
To summarize the conclusions of today's meeting on this: all WireVectors will remain tied to single blocks, but simulation tracing will be indexed by name rather than identity, as Input and Output names will remain the same after synthesis. This access-by-name will also apply to memories. To make the change easier, `.trace` will still allow access by identity, but issue a deprecation warning when it occurs.

We do still need to decide how to handle specifying initial values for registers in synthesis. (By name would make sense, but synthesis has split them into automatically-named single-bit registers.)

abe-k, 2016-05-07:
I have now some apparently-working code that indexes traces by name rather than identity: https://github.com/abe-k/PyRTL/commit/94967aef2b74d9130f1ce68214dcd1d6d306a260. I suspect, though, that there are a lot of untested edge cases it will fail on.

Gamrix, 2016-06-29:
Is there anything that still needs to be done for this issue???

closed 2016-07-08 by Gamrix