UCSBarchlab/PyRTL #138: Optimize breaks __iter__

jemcmahan13, 2016-04-27:
The analysis in `TimingAnalysis` calls `__iter__`. Under certain conditions, this works fine, but hits an error *after* a block has been synthesized and optimized. A pickled block is attached, with the following code run on that block.

[error.zip](https://github.com/UCSBarchlab/PyRTL/files/239543/error.zip)

```python
import pickle
from pyrtl import *
from pyrtl.analysis import *

with open("error.pickle", "rb") as f:
    p = pickle.Unpickler(f)
    b = p.load()  # b is the block

syn_b = synthesize(False, b)  # syn_b is the synthesized version of the block
opt_b = optimize(False, syn_b)  # opt_b is the optimized version of the block

t = TimingAnalysis(b)
print(t.max_freq())  # timing for b

syn_t = TimingAnalysis(syn_b)
print(syn_t.max_freq())  # timing for synthesized block

opt_t = TimingAnalysis(opt_b)
print(opt_t.max_freq())  # timing for optimized block (throws error)
```

```
56.140666052862045
31.668047921356912
Traceback (most recent call last):
  File "opterror.py", line 18, in <module>
    opt_t = TimingAnalysis(opt_b)
  File "/home/joseph/work/PyRTL/pyrtl/analysis/estimate.py", line 155, in __init__
    self._generate_timing_map(gate_delay_funcs)
  File "/home/joseph/work/PyRTL/pyrtl/analysis/estimate.py", line 187, in _generate_timing_map
    for _gate in self.block:  # ordered iteration
  File "/home/joseph/work/PyRTL/pyrtl/core.py", line 329, in __iter__
    find_and_print_loop(self)
  File "/home/joseph/work/PyRTL/pyrtl/helperfuncs.py", line 607, in find_and_print_loop
    loop_data = find_loop(block)
  File "/home/joseph/work/PyRTL/pyrtl/helperfuncs.py", line 549, in find_loop
    result = _check_for_loop(block)
  File "/home/joseph/work/PyRTL/pyrtl/helperfuncs.py", line 540, in _check_for_loop
    wires_left.discard(*net.dests)
TypeError: discard() takes exactly one argument (0 given)
```

One issue is that a loop is apparently encountered in the optimized block when no loops were found before, causing `find_and_print_loop` to be called. The second problem is that the code in `_check_for_loop` attempts to discard wires in `net.dests` for an `@` op, which is defined as having no dests.

jemcmahan13, 2016-05-02:
That change (b440783) has addressed a particular symptom, but the problem still persist that `optimize` seems to be (for the attached block at least) introducing new, errant behaviors into the block. The error has evolved to `Wires used but never driven` on a set of wires; debug indicates that all of these wires were created in the call to `synthesize`. What is strange (as with the original error) is that everything works fine post-synthesis, but breaks post-optimization.

Also, a question on 	b440783: if `@` ops weren't being passed in and causing problems before, what changes post-optimization that make the original error occur? I.e., why is this guard needed?

Gamrix, 2016-05-02:
I know that there is still a big to be fixed. I have narrowed down the problem to `remove_unlistened_nets` (at least that is where it turns into something catchable by `sanity_check`) but its removing 600 Logicnets in a black box pool of 25k nets, and its hard to figure out which ones it is making a mistake on. If you have a simple example where this also happens, I would appreciate it if you would post it here.

jemcmahan13, 2016-05-02:
I tried a couple times to make a small one, but it didn't show up. If I had a smaller example, I would certainly post it.

It's possible there's something wrong in the block itself and the issue is that we have a missing check for some unexpected property.

closed 2016-05-04 by Gamrix