UCSBarchlab/PyRTL #141: Split helperfuncs.py into two files

Gamrix, 2016-04-29:
Currently there is a lot of code to make small circuits. For Example `and_all_bits`, `mux` and `_basic_mult` and helperfuncs is ballooning into a massive 600 lines of code. I am wondering if it would make sense for us to separate this functionality into a `corecircuits.py` file (I'm taking suggestions for names). 

suggested criteria for going into `corecircuits.py`: the code constructs circuitry that is useful for production (not debug).

abe-k, 2016-04-29:
Another way of splitting up `helperfuncs.py`, which I was considering, is to move the implementations of built-in operators (i.e. the `_basic*` functions) into their own file, and leave the rest as-is. These two splits aren't mutually exclusive, of course.

timsherwood, 2016-05-01:
Will this change the import path for these functions?

Gamrix, 2016-05-01:
The plan would be yes for PyRTL internal use. It will not change pyrtl._____ semantics (aka pyrtl.mux will still be pyrtl.mux), but it will change internal function locations. For example helperfuncs.mux will no longer work as it will be in a new file.

commit 1861f898 (2016-06-08): Refactored Helperfuncs into two files. Closes #141
Also have a nonworking optimization that I am currently working on in this commit
closed 2016-06-08 by Gamrix