UCSBarchlab/PyRTL #145: Compile to C to speed simulation

abe-k, 2016-05-11:
I've started working on a class, `CompiledSimulation`, that will convert PyRTL logic to C code, compile it, and import it using `ctypes`. This issue can be a place to discuss concerns, potential features, and performance. For my initial proof-of-concept, I'm not planning to tackle big-integer support, but rather require that all WireVectors fit in 64 bits.

Gamrix, 2016-05-11:
That's totally fine. I think the first thing we want is something to give us a baseline so that we can compare performance. It doesn't need to be complex.

abe-k, 2016-05-11:
I have `CompiledSimulation` working on a copy of the `test_fastsimulation` suite, with the exception of anything more than 64-bit-wide. What sample hardware description would you recommend for performance testing?

timsherwood, 2016-05-11:
If we could run it on the H-machine that would be the most directly useful.  AES is a good one as well.  I think the target it to understand what happens when you run for 10e4 or 10e5 cycles or so.

Gamrix, 2016-05-11:
Also, PyRTL-Research/Performance/sampleBlocks.py has some blocks I have used before for profiling. You might want to test against them. There is a multiply block, AES, as well as two adders.

You will want to set a custom bitwidth for multiply, and since you cannot handle bitwidths over 64, you can't run AES, but they should still be useful.

abe-k, 2016-05-11:
Some preliminary performance results, for a 31-by-31 tree multiplier with 100 simulation steps (times in seconds):

 | FastSimulation | CompiledSimulation
---|---|---
elaboration | 0.26 | 0.26
compilation | 0.42 | 32.4
simulation | 0.46 | 0.0045

I then tested CompiledSimulation with a million simulation steps, and it took 46.4 seconds (plus the fixed elaboration and compilation time), compared to (I assume) over an hour for FastSimulation.

abe-k, 2016-05-11:
I just added support for doing the looping in C (providing a list of inputs at each step and a buffer for the trace of the outputs), and it saves about another 8 seconds for the million-step multiply.

timsherwood, 2016-05-11:
Nice!  Did you compile with -O5?   Given how "weird" the generated code will be (compared to run-of-the-mill C code) I wonder if doing a search over the set of compiler flags might make it even faster?  Assuming that the interface is the same, this is a big big win for doing stuff like H-machine.  Also, I would be curious how it does on something with lots of memories.  We should definitely try it on the H-machine.

abe-k, 2016-05-12:
They say premature optimization is the root of all evil. They are very much correct.

Specifically, the results I gave earlier are with the `-O3` flag, which is GCC's highest (`-O5` is equivalent). I just tried reducing the optimization level. The effect on run time was negligible (less than the variation between runs, which is about a second), but compile time drops all the way down to 1.4 seconds with the no-optimization setting, `-O0`.

Memories should be fast, as they're just stored as arrays. Where is the H-machine, so I can test with it?

abe-k, 2016-05-12:
I've now managed even better results: by making all the variables static (i.e. not exposed in the library), compile time is just over a second, and run time is under 23 seconds for the million steps (under 20 if looping in C). The only downside is that we lose the ability to inspect or trace arbitrary wires and memories—only Outputs are exposed.

At this point, for this test case, CompiledSimulation is faster than FastSimulation for anything more than about a hundred steps.

timsherwood, 2016-05-12:
Awesome.   How many Hz does AES run at now?  Assuming this works on H-machine we can kill of FastSimulation.

Gamrix, 2016-05-12:
@jemcmahan13 Can you give abe access to the H-Machine?

jemcmahan13, 2016-05-12:
There's a version of the block posted on https://github.com/UCSBarchlab/PyRTL/issues/138.

You'll need to add support for larger wirevectors first, however; there are a few wide memories/wires.

Gamrix, 2016-05-13:
I was thinking about the long int support problem, and I think that you could get around it by instead 'synthesizing' wires that are over the max int length into wires that are under the limit. As the user can't observe arbitrary wires, doing this extra synthesis should be rather harmless.

abe-k, 2016-05-13:
I've pushed my CompiledSimulation implementation: https://github.com/abe-k/PyRTL/commit/18f048ca40909a4e9f3c19270fa9473ecbec4408; have a look.

Would you suggest implementing the reduction to 64-bit with a transform pass, or how?

Gamrix, 2016-05-13:
There would need to be a wire vector map akin to synthesis but other than that, I think that transforms would do the trick just fine.

jemcmahan13, 2016-05-13:
If the simulation is over pre-synthesis blocks, won't you have operations that can't be easily broken down? I.e., AES might have a 128-bit multiply. You'd have to either swap in a logical implementation or provide a recursive, composable definition whereby ops can be broken into parts.

Gamrix, 2016-05-13:
Yeah, there will have to be something interesting for subtractions and multiplications in particular, but I bet it wouldn't be that hard to implement.

abe-k, 2016-05-13:
As an experiment, I tried using GCC's built-in support for 128-bit integers (`__attribute__((mode(TI)))`) to run the AES decryption. Preliminary results are that compilation takes 6.3 seconds, but then it can do 20 kilocycles in 0.72 seconds (compare to almost a minute for FastSimulation). The problem is that when 128-bit integers aren't needed (e.g. the 31 by 31 tree multiply), using them slows things down by an order of magnitude, so this is not a real solution.

Gamrix, 2016-05-13:
Also, the other problem is that we can easily have bitwidths over 128 anyways. Other forms of AES use 192 and 256 bits in their ciphers. And don't even get me started on SHA and RSA.

Gamrix, 2016-05-13:
Also, just so that you are aware, I have been working on making fastsim faster.

I have been able to get up to a 10x performance improvement

(FastSimulation is the version currently in simulation.py, FS_execFunction is the new version)
````
C:\Users\John\AppData\Local\Programs\Python\Python27\python.exe C:/Users/John/Documents/PycharmProjects/PyRTL-research/performance/FastsimPerf/fastsimProfiler.py
Timing AdderCase
FastSimulation_5_16 creation x10 execution time: 0.732177860777
FastSimulation_5_16 simulation x100 execution time: 9.62122820822
FS_ExecFunction creation x10 execution time: 0.614756828917
FS_ExecFunction simulation x100 execution time: 1.13154532796
Timing MultiplierCase
FastSimulation_5_16 creation x10 execution time: 48.2084736751
FastSimulation_5_16 simulation x100 execution time: 90.3165466121
FS_ExecFunction creation x10 execution time: 5.94418384067
FS_ExecFunction simulation x100 execution time: 9.88124481539
Timing AESCase
FastSimulation_5_16 creation x10 execution time: 6.89609526474
FastSimulation_5_16 simulation x100 execution time: 60.9185137079
FS_ExecFunction creation x10 execution time: 3.72845847921
FS_ExecFunction simulation x100 execution time: 39.223340945
````

If you want to look at it and test the c version against it, you can find it in `PyRTL-research/performance/FastsimPerf/fastsimProfiler.py`

abe-k, 2016-05-14:
Here's my results using your suite with the 64-bit version of CompiledSimulation:
```
Timing MultiplierCase
CompiledSimulation creation x10 execution time: 3.945386
CompiledSimulation simulation x10000 execution time: 0.27072499999999966
FS_ExecFunction creation x10 execution time: 3.753169999999999
FS_ExecFunction simulation x10000 execution time: 4.846886
```
This is after fixing a small bug in the suite that resulted in doing the square of the number of steps specified for simulation.

Gamrix, 2016-05-14:
Good catch, I was wondering why my numbers were looking so strange.

On another note, I'm interested in whether performance changes if you do something like what I did in : 
```python
if len(net.dests[0]) == self.expected_bitwidth[net.op](net):
    prog.append("    %s = %s" % (result, expr))
else:
    mask = str(net.dests[0].bitmask)
    prog.append('    %s = %s & %s' % (result, mask, expr))
```

In the Python version, I get almost double the performance on the small op heavy benchmarks by doing this optimization (surprise Python is unable to optimize them away). It would be interesting to see if you get a comparable gain in C++

````
Timing AdderCase
FS_ExecFunction creation x10 execution time: 0.567668867554
FS_ExecFunction simulation x2500 execution time: 0.281213307722
FS_ExecFunction_WOLessMasks creation x10 execution time: 0.588316645484
FS_ExecFunction_WOLessMasks simulation x2500 execution time: 0.477788325917
Timing MultiplierCase
FS_ExecFunction creation x10 execution time: 5.24220492947
FS_ExecFunction simulation x2500 execution time: 2.23202360449
FS_ExecFunction_WOLessMasks creation x10 execution time: 4.85786596278
FS_ExecFunction_WOLessMasks simulation x2500 execution time: 3.89547167551
Timing AESCase
FS_ExecFunction creation x10 execution time: 1.81414620378
FS_ExecFunction simulation x2500 execution time: 9.09154659155
FS_ExecFunction_WOLessMasks creation x10 execution time: 1.78446212557
FS_ExecFunction_WOLessMasks simulation x2500 execution time: 9.62286389919
````

You can see how I did this (along with the `self.expected_bitwidth` dict I made) by looking at `FS_NoMasks` in `FastsimPerf/newFastsims.py`.

abe-k, 2016-05-14:
Selectively omitting the mask doesn't seem to make a substantial difference for CompiledSimulation, and it appears that the mask can always be omitted for the multiplier test, so it doesn't seem like it will matter for any of the test cases.

For `expected_bitwidth`, I think you meant to add for the `'*'` case. Also, can't you just get the MemBlock bitwidth for `'m'`?

abe-k, 2016-05-14:
Another big speedup for CompiledSimulation: using `clang` instead of `gcc`. The improvements (for both compile and run time) are small for the tree multiplier (1.6x speedup), but impressive for AES (5x speedup on compilation, 2.9x speedup on run). AES-128 decryption is almost at 100 kHz now, using the native 128-bit integers. Of course, same caveats as before apply: native 128-bit is really slow when it's not needed, and we should do proper big integer support.

(These results are with the latest compiler versions available for Ubuntu 14.04: `clang 3.4-1ubuntu3` and `gcc 4.8.4-2ubuntu1`.)

Gamrix, 2016-05-14:
Yeah, those are good catches. I actually did not implement `m` because I didn't have sanity checking done for the memory inputs. (no reason to make these microoptimizations until I have figured out the major ones.

One thing I noticed is that the AES simulation actually did not improve much when going to `FS_execFunction`. After looking at the generated code, I suspect this is due to the extremely inefficient implementation of `s` that we have. I'll go about seeing if optimizing that will make a big difference.

Gamrix, 2016-05-14:
Python once again shows how incapable it is of optimizing integer operations (not that I am surprised). Optimizing the `s` op code generation gave me a 2x speedup for the AES case (and no change in the other test cases)

````
C:\Users\John\AppData\Local\Programs\Python\Python27\python.exe C:/Users/John/Documents/PycharmProjects/PyRTL-research/performance/FastsimPerf/fastsimProfiler.py
Timing AdderCase
FastSimulation_5_16 creation x10 execution time: 0.731049656575
FastSimulation_5_16 simulation x10000 execution time: 9.86847108228
FS_ExecFunction creation x10 execution time: 0.559028464387
FS_ExecFunction simulation x10000 execution time: 1.33681490313
FS_ExecFunc_FastSplit creation x10 execution time: 0.608749449231
FS_ExecFunc_FastSplit simulation x10000 execution time: 1.16211022596
Timing MultiplierCase
FastSimulation_5_16 creation x10 execution time: 45.9555557288
FastSimulation_5_16 simulation x10000 execution time: 85.237122267
FS_ExecFunction creation x10 execution time: 4.98309909079
FS_ExecFunction simulation x10000 execution time: 8.8891656294
FS_ExecFunc_FastSplit creation x10 execution time: 4.90775515261
FS_ExecFunc_FastSplit simulation x10000 execution time: 8.45575921579
Timing AESCase
FastSimulation_5_16 creation x10 execution time: 6.82635460402
FastSimulation_5_16 simulation x10000 execution time: 56.8092444642
FS_ExecFunction creation x10 execution time: 2.02492797955
FS_ExecFunction simulation x10000 execution time: 36.1211203109
FS_ExecFunc_FastSplit creation x10 execution time: 1.17804375135
FS_ExecFunc_FastSplit simulation x10000 execution time: 17.1729073453
````

abe-k, 2016-05-14:
Interestingly, turning on `-O1` speeds up AES by 1.4x, for only a slight compile time increase, but doing the same for the tree multiplier slows down compilation by several seconds for no gain in runtime speed. This may indeed have something to do with `s`, as clang at `-O0` does not simplify it. What are you doing to speed up `s` in Python (as it may also help in C)?

Gamrix, 2016-05-14:
I group together multiple wires for the commoncase where we are splitting sequential wires

```python 
elif net.op == 's':
                source = self.arg_varname(net.args[0])
                expr = ''
                split_length = 0
                split_start_bit = -2
                split_res_start_bit = -1

                for i, b in enumerate(net.op_param):
                    if b != split_start_bit + split_length:
                        if split_start_bit >= 0:
                            # create a wire
                            bit = '(%d & (%s >> %d))' % ((1 << split_length) - 1, source, split_start_bit)
                            expr += '(%s << %d) |' % (bit, split_res_start_bit)
                        split_length = 1
                        split_start_bit = b
                        split_res_start_bit = i
                    else:
                        split_length += 1
                bit = '(%d & (%s >> %d))' % ((1 << split_length) - 1, source, split_start_bit)
                expr += '(%s << %d)' % (bit, split_res_start_bit)
````
(find it in FS_ExecFunc_FastSplit)

Gamrix, 2016-05-15:
I think I might also do a few more optimizations to `s`, but they will be nowhere as drastic as this one

Gamrix, 2016-05-27:
I have added a mips case to the fast sim profiling cases. It would be interesting to see how well the C code did on them. On another note, I have updated the fastsim in PyRTL with the latest and greatest fastsim I made.

jemcmahan13, 2016-06-11:
Where does this live? I don't see it in your commit record. I want to test out some h-machine programs to compare the compile/runtimes there (and make sure the answers come out the same).

Gamrix, 2016-06-11:
https://github.com/abe-k/PyRTL

Gamrix, 2016-06-11:
As of last time I heard, there are still some big performance penalties if you use any wires with a bitwidth between 64 and 128 (over 128 is not supported yet)

jemcmahan13, 2016-06-12:
Ah --- in that case, I'll need to hold on for BigInt support.

timsherwood, 2016-06-28:
As a first go to bring this feature into the main trunk (which we would like to do shortly), I think we should be able to use a bigint library (making sure to not release such library with our code or in any other way inherit their licence).  Such bigints should only be used for those wirevectors that require it though for performance.  Thoughts?

Gamrix, 2016-07-22:
We should work on getting a version without big int support into PyRTL soon. Even if we cannot run everything, it would still be useful

Gamrix, 2016-07-28:
Update: I have pulled the latest version of CompiledSim into Development (I have also made a few edits to the code as well). 

Currently, no unittests are run for CompiledSim. There will need to be some small modifications to make CompiledSim unittests run on TravisCI while still being skipped locally (I don't want a lot of complicated dependencies, especially with the quirks of CompiledSim  to be something we have to make all devs deal with).

@abe-k @timsherwood

Gamrix, 2016-07-29:
When running it today, I got the following error.

```
Traceback (most recent call last):
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/processors/edit_distance/edit_dist_with_analysis.py", line 166, in <module>
    gather_stuck_bit_info(test_dicts, ins)
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/processors/edit_distance/edit_dist_with_analysis.py", line 137, in gather_stuck_bit_info
    wire, wire_bit, change_count = test_stuck_bit_compiled_sim(test_dicts, stuck_type)
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/processors/edit_distance/edit_dist_with_analysis.py", line 116, in test_stuck_bit_compiled_sim
    sim = pyrtl.CompiledSimulation()
  File "C:\Users\John\Documents\PycharmProjects\PyRTL\pyrtl\compilesim.py", line 56, in __init__
    self._setup_dll()
  File "C:\Users\John\Documents\PycharmProjects\PyRTL\pyrtl\compilesim.py", line 64, in _setup_dll
    self._dll = ctypes.CDLL(path.join(self._dir, 'pyrtlsim.so'))
  File "C:\Users\John\AppData\Local\Python\Python34\lib\ctypes\__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 1114] A dynamic link library (DLL) initialization routine failed
```
I am going to suround the code with a `try/except` block that will retry generation of the code once for now.

Gamrix, 2016-07-29:
Ok, that workaround did not work. The retry threw the same error. Also it threw it around the same time as well, after around 50 initializations of the CompiledSim.

Gamrix, 2016-07-29:
OK, I think I have fixed it. I have added the ability to unload the dll to `CompiledSim` and it is no longer throwing the error. I haven't checked the validity of the Linux dll closing code though.

abe-k, 2016-10-29:
I am very close to having CompiledSimulation working with arbitrary-bitwidth wires. However, it fails with errors on many of the unit tests, because it does not support tracing non-Output wires. While it seems that there was a consensus that simulations did not need to support this, SimulationTrace assumes that they do, and changing it breaks all the tests and examples. Do we want to make lots of changes to the tests (and possibly to the general behavior of tracing), or do we want to accept the substantial performance hit in CompiledSimulation of supporting inspection?

Gamrix, 2016-10-29:
I'm interested in how much it would cost us to support the amount of tracing `FastSim` currently offers (inputs, outputs, and registers). If that's not large, I would be in favor of parity with `FastSim`, otherwise, I would be fine with making compiled sim skip some of those tests. ( and maybe make a few additional tests in response)

abe-k, 2016-10-29:
If I implement it by making the C variables global, as I was thinking, we get about a 1.6x to 1.8x slowdown, plus the overhead of fetching them if queried, based on what speedup I got by making the opposite change in an early version of CompiledSimulation. (It would also slow down compilation slightly.)

If we instead have a list of WireVectors known at compile-time to be traced, we can pretend they are outputs and avoid having to make arbitrary WireVectors accessible. This would require additional memory (plus a little time) regardless of whether the WireVectors were actually traced. (A complicated design could easily have a thousand potentially-traced wires, and running that for a million steps at once would take an extra 8 GB.)

For the specific case of Inputs, I can add tracing support with no overhead, since their values are already known to the Python code, so I'll do that, at least. For the rest, what seems worthwhile? Requiring `probe` on anything internal (and rewriting tests), accepting either the slowdown or memory requirements, or some other approach I didn't consider?

Gamrix, 2016-10-30:
I would say that it's fine to require probing on anything internal. If someone wants more debugging flexibility, they shouldn't be using compiledsim

abe-k, 2016-10-30:
Alright, I have made a copy of the tests. I've gotten almost all of them passing, and for the remaining two I'm not sure whether to change the code or remove the tests:
* `test_rom_out_of_range_error` would require handling of memories which do not contain entries at every valid address, which would be a complication and performance hit.
* `test_input_out_of_bitwidth` would be easier to fix (just a negligible performance hit on the input handling), but it seems like the sort of thing people should get working _before_ they use CompiledSimulation.

By the way, here are the current results using a slight modification of your `fastsimProfiler`:
```
Timing AdderCase
FastSimulation creation x10 execution time: 0.3320920000001024
FastSimulation simulation x10000 execution time: 0.5044849999999315
CompiledSimulation creation x10 execution time: 1.9407400000000052
CompiledSimulation simulation x10000 execution time: 0.05339899999989939
Timing MultiplierCase
FastSimulation creation x10 execution time: 3.1902430000000095
FastSimulation simulation x10000 execution time: 4.170219000000088
CompiledSimulation creation x10 execution time: 9.16256999999996
CompiledSimulation simulation x10000 execution time: 0.15981799999963187
Timing AESCase
FastSimulation creation x10 execution time: 0.6752329999999347
FastSimulation simulation x10000 execution time: 10.285472000000027
CompiledSimulation creation x10 execution time: 9.371462000000065
CompiledSimulation simulation x10000 execution time: 0.09213899999986097
Timing MIPSCase
Warning: for barrel shifter, the shift distance wirevector has bits that are not used in the barrel shifter
FastSimulation creation x10 execution time: 0.6595440000000963
FastSimulation simulation x10000 execution time: 2.066139999999905
CompiledSimulation creation x10 execution time: 8.620217000000139
CompiledSimulation simulation x10000 execution time: 0.05490800000006857
```
I suspect a decent bit of the time is spent in Python-side overhead, so I might try making some tighter benchmarks.

Gamrix, 2016-10-30:
Yeah, I don't think we need support for sparse Registers. However, I think a rudimentary check for the input vals would be very useful. People should have everything working before CompiledSim, but the honest truth is that people aren't always so organized. Even a simple check on the C side would be good enough

abe-k, 2016-10-31:
I've pushed my changes to [abe-k/PyRTL](https://github.com/abe-k/PyRTL).

Note that I've switched back to GCC, because I was having issues with clang and `-mcmodel=medium`. I haven't actually checked if the switch is worthwhile.

There are still some hacks (like removing from `SimulationTrace` any untraceable wires), and probably tons of bugs in untested parts of the code (especially the higher-bitwidth support), but it passes the tests it's supposed to, and I think it's basically useable. I encourage anyone reading this to try it out and tell me what doesn't work.

timsherwood, 2018-03-08:
BTW, Joseph has a big design now that he needs to run for many many cycles.  You might ping him about that if you want a stress test for the compiled design.  If sim-compiled is stable on the tests I am happy to merge it into development too.

commit a3c84178 (2018-11-10): CompiledSim ready to go, fixes #145
closed 2018-11-10 by timsherwood