UCSBarchlab/PyRTL #146: Fastsim Step doesn't work with wire names

Gamrix, 2016-05-13:
I tried passing wirevector names into fastsim, and it gave me the following error.

````
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/performance/FastsimPerf/fastsimProfiler.py", line 123, in sim_all
    sim_case(prof_case)
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/performance/FastsimPerf/fastsimProfiler.py", line 117, in sim_case
    prof.run_sim()
  File "C:/Users/John/Documents/PycharmProjects/PyRTL-research/performance/FastsimPerf/fastsimProfiler.py", line 28, in run_sim
    self.sim.step({wire.name: val[cycle] for wire, val in zip(inwires, invals)})
  File "C:\Users\John\Documents\PycharmProjects\PyRTL-research\performance\FastsimPerf\fastsim_5_2016.py", line 96, in step
    self.context[self.varname(wire)] = wire.bitmask & value
AttributeError: 'str' object has no attribute 'bitmask'
````

Gamrix, 2016-06-07:
Similarly, Simulation.inspect() does not work with names.

jemcmahan13, 2016-06-08:
No one answered my slack question on this, so, to be clear: we've decided previously in meetings that the simulation-time interface for *all* simulators would operate *only* off of names; is that correct? It solves some issues of simulation post-synthesis, and a user can always change `inspect(wire)` into `inspect(wire.name)`.

Gamrix, 2016-07-07:
Yes, that is correct. Also the simulators have been updated so that this works.

closed 2016-07-07 by Gamrix