UCSBarchlab/PyRTL #148: Assignments to "name" break FastSimulation

jemcmahan13, 2016-06-02:
Simulation and FastSimulation should present the same interface, but some code that works with the former will not work with the latter. Specifically, if an elaboration program assigns to `.name` of a `WireVector`, FastSimulation encounters a `KeyError`.

```
from pyrtl import *

inw = Input(8)
outw = Output(8)
w = WireVector(8)
w <<= inw
outw <<= w

w.name = "Instr"

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)  # this line causes an error
sim = Simulation(tracer=sim_trace)  # with previous line commented out, this works fine
```

The problem seems to be because `FastSimulation` relies on the wire names in the `Tracer` object, which are not updated when `name` is assigned to. It seems like the options are either to 1) make `name` updates do all related bookkeeping to keep stuff from breaking (are there other places?); or 2) make `name` immutable.

jemcmahan13, 2016-06-02:
Getting a new error (under Python 2.7) with the most recent commit:

```
Traceback (most recent call last):
  File "main.py", line 119, in <module>
    sim = FastSimulation(tracer=sim_trace)
  File "/home/joseph/work/PyRTL/pyrtl/simulation.py", line 310, in __init__
    self._initialize(register_value_map, memory_value_map)
  File "/home/joseph/work/PyRTL/pyrtl/simulation.py", line 344, in _initialize
    logic_creator = compile(s, '<string>', 'exec')
  File "<string>", line 13479
SyntaxError: can't assign to operator
```

jemcmahan13, 2016-06-02:
Attaching pickle of current circuit, as requested for @Gamrix .

[error.pickle.zip](https://github.com/UCSBarchlab/PyRTL/files/296597/error.pickle.zip)

Gamrix, 2016-06-02:
Rename your `Heap-xxx` variables so that they are valid python identifiers. I will talk with others about a more permanent solution soon.

timsherwood, 2016-06-03:
I think that the opposite question is better — is there compelling reason to require them to be?  It is conveinent to have spaces, dashes, parens,  and other symbols in the names, especially as we require them to be unique even when generated.

abe-k, 2016-06-03:
Agreed. The only reason to require valid Python identifiers that I know of is for a trivial convenience in the `FastSimulation` implementation. If for some reason we do decide to require valid identifiers as names, we should at a minimum fail early with a clear error message, but I think that's more work than fixing the simulator.

Gamrix, 2016-06-03:
I was just worried that other output formats will have to have their names garbled in order to work (eg: #147 VCD files).

timsherwood, 2016-06-03:
It might make sense to add a good name sanitizer to the helperfuncs for these types of problems?

commit 1008e24b (2016-06-02): Added more wire name consistency
Fixes #148
Added additional unittests to check that wire name changes are actaully reflected in the system
closed 2016-06-08 by Gamrix