UCSBarchlab/PyRTL #149: block_to_svg() available in 0.8.2?

sg2, 2016-06-04:
In the documentation [1], the method `pyrtl.inputoutput.block_to_svg()` is listed; however, I don't see that as an option after running example1-combologic.py. And when I try to run the method, I get: 

```
In [15]: pyrtl.inputoutput.block_to_svg()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-15-3060d3f9b679> in <module>()
----> 1 pyrtl.inputoutput.block_to_svg()

AttributeError: 'module' object has no attribute 'block_to_svg'
```

Am I doing something incorrectly? I can successfully call other methods of `pyrtl.inputoutput`.

[1] http://pyrtl.readthedocs.io/en/latest/inputoutput.html

Gamrix, 2016-06-04:
Sorry, yeah, we need to make a new release. The docs is based on the latest build in PyRTL Development.

timsherwood, 2016-06-28:
New release should fix this issue.  Sorry for the delay

closed 2016-06-28 by timsherwood