UCSBarchlab/PyRTL #152: Use NameSanitizer in other places

Gamrix, 2016-06-08:
Recently I have built a generic `NameSanitizer` as well as base classes for PyRTL.
This is out of the work in issue #148

Places where the old code should be replaced are:
- [ ] Verilog Output
- [ ] Global Temporary Names (names made by normal `WireVector` instantiations)
- [ ] Probe Names (use the `NameIndexer` to generate probe indexes)

If there are places that I have forgotten please mention them below.

jemcmahan13, 2016-06-08:
The Name Sanitizer commit (c90d2bde683521aeeb) broke my code. The error seems to be coming from the compiled code, which makes it hard to place where something went wrong.

```
Traceback (most recent call last):
  File "main.py", line 189, in <module>
    sim.step({inval: feed_value})
  File "/home/joseph/work/PyRTL/pyrtl/simulation.py", line 379, in step
    self.regs, self.outs, mem_writes = self.sim_func(ins)
  File "<string>", line 98, in sim_func
KeyError: u'Heap-raddr'
```
(Different wires, all with names that need to be sanitized, show up as the argument of the KeyError.) Is there a place in the compiled code that uses the unsanitized names? Not sure why else it would throw a KeyError on the original name string.

Gamrix, 2016-07-13:
Yeah, the `AssertIndexer` and `ProbeIndexer` classes seem like overkill. Could we just use `NameIndexer` instances and just put the rest of the code in the `probe` and `rtl_assert` functions? the `AssertIndexer` is particularly egregious because it basically looks like a copy of `NameIndexer`.

closed 2016-07-20 by Gamrix