UCSBarchlab/PyRTL #153: Error in VCD output -- Problem with negative values?

jemcmahan13, 2016-06-08:
Outputting a wire's value for a VCD trace uses the Python `bin` function and removes the leading 0 (e.g. `0b10` becomes `b10`). I'm getting some issues where the leading 0 is not removed, which I think is due to negative values at simulation time. Python's `bin` gets super weird (or "helpful") with negatives, because Python integers have no bitwidth --- e.g., `bin(-3)` is just `-0b11`. Taking `bin(-3)[1:]` gives a VCD output of `0b11` (instead of `b11`), which causes errors in the waveform viewer that corrupt the trace.

Fixing the VCD problem seems easy, but there might be other issues with how negative values are handled. I think Python's semantics for bit-operations on negative values will work the way we expect, but I'm not sure all cases are covered. How is the compiled C code handling negatives? An arbitrary-precision negative number has infinite leading 1's, so optimization attempts to remove the bitmasking operations could cause errors if negative values are present.

PyRTL's trace renderer will output a negative sign, like `-0x1`; is that what we want?

Here's an example that creates a VCD file with a bad value in it and also shows the negative values in the terminal trace.

```Python
from pyrtl import *

inw = Input(8, "inw")
outw = Output(8)

val = inw - 1
outw <<= val
val.name = "value"

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)

sim.step({inw: 1})
sim.step({inw: 0})

with open("bad.vcd", 'w') as f:
    sim_trace.print_vcd(f)

sim_trace.render_trace()
```

jemcmahan13, 2016-06-08:
Okay, I'm going to go ahead and say we've got some weirdness manifesting in FastSimulation because of negative values.

First consider the following, which breaks the trace renderer:
```Python
from pyrtl import *

inw = Input(8, "inw")
outw = Output(8, name="outw")

val = (inw - 1)[-1]
outw <<= val + 1
val.name = "value"

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)

sim.step({inw: 1})
sim.step({inw: 0})

sim_trace.render_trace()
```
This ends up with:
```
File "/home/joseph/work/PyRTL/pyrtl/simulation.py", line 628, in _render_val_with_prev
    out = pretty_map[(self.prior_val, current_val)]
KeyError: (0, -1)
```
This appears to be because it's trying to render a 1-bit wire (which is rendered as a line, not a number), but has no way to render a value `-1`.

Now, for more weirdness, we can actually make a small change to get rid of the error. By moving the select operation `[-1]` to a new line (with a new wire that isn't being rendered), the old error goes away and we get a valid trace rendered. That program looks like this:

```Python
from pyrtl import *

inw = Input(8, "inw")
outw = Output(8, name="outw")

val = (inw - 1)
val2 = val[-1]
outw <<= val2 + 1
val.name = "value"

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)

sim.step({inw: 1})
sim.step({inw: 0})

sim_trace.render_trace()
```
But the values are wonky because of the negative wire-value semantics; taking the MSB of the wire with value `-1` results in a one-bit wire with value `-1` (which was breaking the tracer). Unfortunately it also manifests when using the value, as shown in the add operation, which adds the one-bit wire to `1` and gets `0` (instead of `2`) as a result. Terminal output copied here:
```
  inw 0x1  ╳0x0 

 outw 0x1  ╳0x0 

value 0x0  ╳-0x1
```
In the second cycle, when the input is `0`, `value` has `-1` and the output, which adds a one-bit wire to a constant `1`, has `0`.

If you use normal `Simulation`, everything's fine.

jemcmahan13, 2016-06-08:
For the curious, I found this because my ALU tests were failing on less-than. Instead of outputting a 1, it was outputting 0xFFFFFFFF. The following shows how the negative weirdness can lead to a "one bit" wire taking a value larger than 1.

```Python
from pyrtl import *

inw = Input(8, "inw")
outw = Output(8, name="outw")
val = WireVector(len(inw)+1)  # subtract out, so need an extra bit for carry to examine sign later

val <<= (inw - 1)[-1]  # subtract and take just the carry-out to see if the result was negative
outw <<= val[:8]  # truncate to the correct width

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)

sim.step({inw: 1})
sim.step({inw: 0})

sim_trace.render_trace()
```

```
 inw 0x1  ╳0x0 

outw 0x0  ╳0xff
```

closed 2016-06-08 by Gamrix

Gamrix, 2016-06-08:
Yeah, the idea is that fastsim should never have a negative value. Everything breaks as soon as one gets in (if it is not masked to make it positive again). The other problem is that I try to reduce masking in order to get additional speed boosts. In some cases this is close to a 2x speedup by itself (after I made the code stay inside of a function instead of the global scope).

jemcmahan13, 2016-06-08:
Is there a suite of tests we can use as additional optimizations are added to simulations to make sure the semantics haven't been altered? I imagine we'll want that as well to test the C simulation.

Gamrix, 2016-06-08:
I was trying to build one in PyRTL-Research in addition to the basic sim unittests. That was also part of the reason why I wanted the H-Machine sim.

jemcmahan13, 2016-06-09:
I think that's a good idea for some more complicated, niche tests that do weird things. You'll probably still need a suite of other stuff too, though. Once the spec stabilizes this week I can forward the circuit with some basic test programs (which cover most behavior) and we can add more complicated programs if you'd like as the compiler stabilizes.

commit 1b0b0399 (2016-06-08): Always mask subtractions as they can make negative values. Fixes #153
Added unittest to check for subtraction 'overflow'