UCSBarchlab/PyRTL #157: memory port limitations

timsherwood, 2016-06-28:
Building memories with more than 2 read and 1 write port may be problematic from an FPGA synthesis standpoint (2r/2w might be possible too but we need to check).  Suggestion: Add a new parameter to the memory classes that limits the number of ports (throwing an error when more are created) unless explicitly set otherwise.  E.g. max_read_ports=2, max_write_ports=1, where an argument of None indicates there are no limits.  Note: may break some prior code.

Gamrix, 2016-07-01:
Does anyone have any suggestions as to how to prevent legacy code from breaking (maybe have a `lega cy mode` flag). Or maybe it doesn't matter at all....

timsherwood, 2016-07-01:
I would say just break it -- the less flags the better (we are still pre 1.0)  After that we can start worrying about backward compatibility.

closed 2016-07-08 by Gamrix