UCSBarchlab/PyRTL #163: Fix Output to Verilog for Sync Mems

masoug, 2016-06-30:
When PyRTL outputs verilog for sync reads from mems, it uses assign statements: see here https://github.com/UCSBarchlab/PyRTL/blob/development/pyrtl/inputoutput.py#L672 for details.

Unfortunately, the synthesis tools are not smart enough to realize that the addresses for those memory reads are registers. Hence they refuse to synthesize those elements as BRAM and instead tries to convert them to memory LUTS. The solution to this was to put memory reads under an 'always' block and make the mem output wires verilog regs. Doing so does not change the semantics of the logic, it just makes it obvious to the synthesizer that the memory reads are to be done synchronously.

For more details on how the synthesis tools infer block ram, see the Xilinx design guide on page 85: http://www.xilinx.com/itp/xilinx10/books/docs/sim/sim.pdf#page86

Gamrix, 2016-07-01:
Could you post some example code?

masoug, 2016-07-01:
the design guide i linked above is the "ground truth" for Xilinx's synthesis tools. It contains sample verilog code that for-sure converts to brams.

masoug, 2016-07-21:
@Gamrix please reopen

I found another way to specify synchronous memories in the same semantic form as PyRTL. Please see https://github.com/UCSBarchlab/PyRTL/issues/199 for details. The verilog output generator needs to be changed to use blocking assignments for synchronous memory addressing.

closed 2017-05-17 by masoug