UCSBarchlab/PyRTL #171: Restrict MemIndexed

Gamrix, 2016-07-06:
After a talk with Sherwood, we decided that three changes should be made to `MemIndexed`

- Reading and writing the `MemIndexed.name` should throw a descriptive error stating what 
  `MemIndexed` is. As it is a temporary object, it should not have a name
- `MemIndexed` objects should be consumed. After using them for one operation, they should 
   throw an error if a person should use them again 
   (be very careful around conditional operations!!!!!)
- `MemIndexed` should no longer be called `_MemIndexed` as we are exposing it to users

MemIndexed Name Error:
```
MemIndexed is a temporary object and therefore doesn't have a name
```

MemIndexed Reassign Error String:
```
MemIndexed can only be used once. If you are trying to use the result of a read in two different locations
please use pyrtl.aswires() to convert this MemIndexed into a WireVector. If you were intending to make
multiple ports with this MemIndexed assignment (not likely), you must explicitly make multiple ports. 
```

jemcmahan13, 2016-07-06:
I'm unclear how this consumption would work. Would the following be allowed?

```python
x = mem[addr]
y <<= x
z <<= x
```

Once it's clear the object is being used as a read, why not transparently change it into a WireVector?

Gamrix, 2016-07-06:
That code would not be allowed.
The problem with transparently changing it to a `WireVector` is that it would have to stick some additional logic between `x` and `y` on the third line (aka add logic after the fact). 

Another option is that we can transparently change `x` into a `"WireVector"` on line 2 prior to the assignment. (this could actually fix one of the other bugs that we have with memories)

~~The other option is to always have these memory reads always return a WireVector~~ Actually the way Memory writes occur means that this cannot work properly.

Gamrix, 2016-07-07:
~~Actually after thinking about it, all of the alternatives would actually require a ton of very hacky code to implement, as the MemIndexed object is not always the class getting the function call when it is accessed~~

Never mind that. There can be rather simple changes made to as_wire to accommodate this case.

timsherwood, 2016-07-07:
I am not sure how this would work, can you explain?  Also, one other crazier option is for the MemIndexed object transmute itself into a wirevector when it is read from (you can do that in python by setting self.__class__).  I am not suggesting that we should do that though!

Gamrix, 2016-07-07:
I'm thinking of the following in `as_wires(val)`
```
    elif isinstance(val, _MemIndexed):
        # covert to a memory read when the value is actually used
        if val.wire is None:
            val.wire =  as_wires(val.mem._readaccess(val.index), bitwidth, truncating, block)
        return val.wire
```

Gamrix, 2016-07-07:
I think that this implementation would make `_MemIndexed` act more in line with what users expect: like a normal WireVector, sans the ability to read the name initially. Some extra work can be done to allow the user to define a name if we really want it.

timsherwood, 2016-07-07:
I see, you cache it back into the MemIndexed.  However, you need to also need to check that when you write to the memindexed and wire is defined already that it is an error?  Also, I need to think about what happens with conditional updates, but that seems to make sense.

Gamrix, 2016-07-07:
Of course there will be more code to check those conditions. I was just providing the basic jist.

Gamrix, 2016-07-08:
So the list of tasks are:
- make the `as_wires()` change as described above
- make accesses changes to a `_MemIndexed.name` affect the underlying wire, and make the wire if neeeded.
- Keep `_MemIndexed` private.
- write unittests
  - check a memory access using a  `_MemIndexed` as an address works
  - check that `_MemIndexed` works properly with conditional assignment
  - check that using a `_MemIndexed` multiple times does not result in multiple read ports being built.
  - check that names work as intended.
  - any other unittests one can think of

Gamrix, 2016-07-08:
The only way I was able to make both a read and a write from a single `_MemIndexed` is as such.

```python
import pyrtl
a = pyrtl.Input(1)
c = pyrtl.Output(1)
mem = pyrtl.MemBlock(1, 1)

mem_i = mem[a]
mem_i_copy = mem_i   # save off a copy so that I can do a mem_read
mem_i <<= a
mem[a] = mem_i
c <<= mem_i_copy
```

Gamrix, 2016-07-13:
Fixed in pull request #183

closed 2016-07-13 by Gamrix