UCSBarchlab/PyRTL #200: MemBlock Asynchronous=True Does not Work

masoug, 2016-07-20:
Given this pyrtl mem test code:
```
import sys
import pyrtl
from pyrtl import *

# *NOTE* Requires memory sanity check to be disabled!

mem_direct = MemBlock(bitwidth=32, addrwidth=8, name='mem_direct')
mem_combo = MemBlock(bitwidth=32, addrwidth=8, name='mem_combo')
mem_register = MemBlock(bitwidth=32, addrwidth=8, name='mem_register', asynchronous=True)

WE = MemBlock.EnabledWrite
we = Input(2, 'we')

combo_input1 = Input(8, "combo_input1")
combo_input2 = Input(8, "combo_input2")
direct_addr = Input(8, "direct_addr")
write_data = Input(32, "write_data")

mem_direct_out = Output(32, "mem_direct_output")
mem_combo_out = Output(32, "mem_combo_output")
mem_register_out = Output(32, "mem_register_output")

# address memory directly from input
mem_direct_out <<= mem_direct[direct_addr]
mem_direct[direct_addr] <<= WE(write_data, we == 0)

# address memory from combo logic
mem_combo_out <<= mem_combo[(combo_input1+combo_input2)[0:8]]
mem_combo[(combo_input1+combo_input2)[0:8]] <<= WE(write_data, we == 1)

# address memory from reg
addr_register = Register(8, "addr_register")
addr_register.next <<= direct_addr
mem_register_out <<= mem_register[addr_register]
mem_register[addr_register] <<= WE(write_data, we == 2)

optimize()

# compile to verilog
with open('hardware.v', 'w') as fp:
    OutputToVerilog(fp)
    
# run pyrtl simulation
simvals = {
    'we':               "000000000333333333111111111333333333222222222333333333",
    'combo_input1':     "000000000000000000123456789123456789000000000000000000",
    'combo_input2':     "000000000000000000123456789123456789000000000000000000",
    'direct_addr':      "123456789123456789000000000000000000123456789123456789",
    'write_data':       "123456789000000000123456789000000000123456789000000000"
}

sim_trace = pyrtl.SimulationTrace()
sim = pyrtl.Simulation(tracer=sim_trace)
for cycle in range(len(simvals['we'])):
    sim.step({k: int(v[cycle]) for k, v in simvals.items()})

with open('trace.vcd', 'w') as fp:
    sim_trace.print_vcd(fp)
```

yields this verilog output: 

```
// Generated automatically via PyRTL
// As one initial test of synthesis, map to FPGA with:
//   yosys -p "synth_xilinx -top toplevel" thisfile.v

module toplevel(combo_input2, we, direct_addr, write_data, mem_direct_output, combo_input1, mem_combo_output, mem_register_output, clk);
    input[1:0] we;
    input[7:0] combo_input2;
    input[7:0] combo_input1;
    input[7:0] direct_addr;
    input[31:0] write_data;
    input clk;
    output[31:0] mem_direct_output;
    output[31:0] mem_combo_output;
    output[31:0] mem_register_output;

    reg[7:0] addr_register;
    reg[31:0] tmp6;
    wire const_0_0;
    wire[1:0] const_4_2;
    wire tmp3;
    wire[8:0] tmp4;
    wire const_1_0;
    wire tmp1;
    wire[1:0] tmp2;
    wire[7:0] tmp5;
    reg[31:0] tmp0;
    wire tmp13;
    wire const_2_1;
    wire tmp11;
    wire const_3_0;
    wire[1:0] tmp10;
    reg[31:0] tmp12;

    reg[31:0] mem_1[255:0];
    reg[31:0] mem_2[255:0];
    reg[31:0] mem_0[255:0];

    assign const_2_1 = 1;
    assign const_1_0 = 0;
    assign const_3_0 = 0;
    assign const_4_2 = 2;
    assign const_0_0 = 0;
    assign tmp10 = {tmp1, const_2_1};
    assign tmp4 = combo_input1 + combo_input2;
    assign tmp5 = {tmp4[7], tmp4[6], tmp4[5], tmp4[4], tmp4[3], tmp4[2], tmp4[1], tmp4[0]};
    always @( posedge clk )
    begin
        tmp12 <= mem_2[addr_register];
    end
    assign mem_register_output = tmp12;
    assign mem_combo_output = tmp6;
    assign mem_direct_output = tmp0;
    always @( posedge clk )
    begin
        tmp6 <= mem_1[tmp5];
    end
    assign tmp1 = {const_1_0};
    assign tmp2 = {tmp1, const_0_0};
    assign tmp11 = we == tmp10;
    assign tmp13 = we == const_4_2;
    assign tmp3 = we == tmp2;
    always @( posedge clk )
    begin
        tmp0 <= mem_0[direct_addr];
    end

    always @( posedge clk )
    begin
        if (tmp3) begin
                mem_0[direct_addr] <= write_data;
        end
        addr_register <= direct_addr;
        if (tmp11) begin
                mem_1[tmp5] <= write_data;
        end
        if (tmp13) begin
                mem_2[addr_register] <= write_data;
        end
    end
endmodule
```

notice that although i set one of the memories to be asynchronous, all of the verilog output memories are still under posedge blocks. To my knowledge, async mems in verilog are not supposed to be under posedge blocks.

Gamrix, 2016-07-20:
so what should the memory declaration look like instead?

masoug, 2016-07-20:
https://github.com/UCSBarchlab/PyRTL/issues/199

jemcmahan13, 2016-07-21:
So it seems like we can keep the current PyRTL synchronous memory semantics and just change the verilog output. However, Sammy's experiments in #199 suggest that those semantics might actually be more limited than they need to be: e.g., a wire can be the result of some combinational logic and still address a synchronous memory block. Barring additional contradictory evidence, it looks like making an asynchronous memory synchronous just required putting the address into a `reg` with a blocking assignment and using that `reg` to address the memory.

@timsherwood

masoug, 2016-07-21:
@jemcmahan13 @timsherwood sorry to keep changing things, but I updated #199 to my latest findings:
```verilog
module v_rams_01 (clk, en, we, addr, di, data_out);
    input  clk;
    input  we;
    input  en;
    input  [5:0] addr;
    input  [15:0] di;
    output [15:0] data_out;
    reg    [15:0] RAM [63:0];
    reg [5:0] addr_reg;

    assign data_out = RAM[addr_reg];

    always @(posedge clk)
    begin
        if (en)
        begin
            if (we)
              RAM[addr] <= di;
        end
        addr_reg <= addr;
    end
endmodule
```
^^^ this synthesizes to synchronous block ram. Note that only the read address needs to be a register. If `RAM[addr] <= di;` was `RAM[addr_reg] <= di;` this will **not** turn into block ram because of the non-blocking assignments. However, if you did `addr_reg = addr` and *then* `RAM[addr_reg] <= di;` that will become block ram. The two approaches listed in this comment have identical timing.

timsherwood, 2017-05-13:
Is there really still a bug here?  In other words, it if fine if a BRAM is inferred even if it is "async" is specified.  As long as the verilog is behaviorally the same?

masoug, 2017-05-16:
Yea originally I was confused by something like `MemBlock(bitwidth=32, addrwidth=16, name='mem', asynchronous=False)` not turning into BRAMs in vivado, but the semantics are still the same i.e. timing is okay.

I was just not sure how to force vivado to make brams when I tried to synth hmachine sometimes. If hardware like hmachine can synth brams with the current pyrtl version right now then we can close this issue.

closed 2017-05-17 by masoug