UCSBarchlab/PyRTL #203: Make print_trace more readable

Gamrix, 2016-07-20:
When print trace is printing the result of multi_digit values, there should be spaces between each individual number
Also, the values coming out of the print trace should be aligned so that they are easier to read.

Finally, there should be a way to specifiy printing a different base. When printing, also print a first line stating which base all the values are in (binary, decimal, etc.)

Gamrix, 2016-07-21:
Also, I would recommend that you right align the values
aka
```
  1
 23

not: 
 1
 23
```

nvandervoort, 2016-07-21:
@Gamrix what about the wire names? Should it look like:
```
in1     1
in2    23
out    24

Gamrix, 2016-07-21:
yeah, wire named should be left justified

nvandervoort, 2016-07-21:
Should these formatting changes only affect the print when the outputs are more than one digit? If not, numerous tests will break.

Gamrix, 2016-07-22:
By default, they should add spaces for one digit numbers. I wouldn't be opposed to a optional `compact` parameter that ommits spaces, as long as it defaults to `False`.

nvandervoort, 2016-07-22:
So you want the exact opposite of what I proposed? That will break about 18 tests in conditional and simulation...

Gamrix, 2016-07-22:
You're welcome to add the optional parameter to the function that validates all those unittests (it shouln't be a change in all 18 places).

nvandervoort, 2016-07-22:
But the tests in those places assume the answer *will* be compact, so setting a parameter _compact_'s default to _False_ *would* change the result in all of those places.

Gamrix, 2016-07-22:
Yes, I am aware of that, sometimes making the correct change is painful. 18 lines of straightforward changes is not that bad.

nvandervoort, 2016-07-22:
Okay, I was just making sure you got what I meant because you said there wouldn't be changes in 18 places. 
On another note, how does this look?
```
                --- Values in base 8 ---
(Probe-0: tmp693)    57   614   261 16010   310    22   221    14  1227   267
adder_probe         117   654   321 16050   350    62   261    54  1267   327
in1                  57    41     3   212    12     3     5     3   335     3
in2                   1    14    73    64    24     6    35     4     3    75
out                 136    30   142    20   220    44    42    30    56   156
select_probe         13     3    14     2    22     4     4     3     5    15
std_probe            57   614   261 16010   310    22   221    14  1227   267
stdout_probe         57   614   261 16010   310    22   221    14  1227   267
```

Gamrix, 2016-07-22:
I'm fine with either of the choices.
The output looks great. (just make sure that the default is either `dec` or `hex`)

closed 2016-07-27 by Gamrix