UCSBarchlab/PyRTL #218: Optimization interferes with sim.inspect

jemcmahan13, 2016-08-11:
With optimization on, some wires that are given names in the program can no longer be accessed during simulation with `sim.inspect`.

Gamrix, 2016-08-11:
Yeah, those wires are likely removed or changed. I don't know what could really be done about this, except to not run optimization.

jemcmahan13, 2016-08-12:
That's a clear answer, and works, but the question is whether optimization _should_ have that behavior. If your system depends on being able to access named wires at runtime, you can never run the optimized version.

Your answer implies that it's impossible to have optimization leave certain wires alone? Or track which wires the original wires correspond to?

Gamrix, 2016-08-12:
Output wires are never removed. If you seek to get a value, the proper way to do it is to use an output wire. (everything else is not guaranteed at this time).

Gamrix, 2016-08-12:
this can be done through probe. Another way this could be done is through a script that connects all named wires to an output.

jemcmahan13, 2016-08-12:
From what I can see, probe does not solve the problem.

```
joseph@Daedalus:~/work/h-machine/multicycle$ python main.py tests/stack.cem.bin --stats
(Probe-0: heap_we)     No call info found for wire: use set_debug_mode() to provide more information
Traceback (most recent call last):
  File "main.py", line 503, in <module>
    if sim.inspect("heap_we"):
  File "/home/joseph/work/PyRTL/pyrtl/simulation.py", line 440, in inspect
    return self.context[self._to_name(w)]
KeyError: 'heap_we'
```

As you can see from the printout, `heap_we` has a probe on it, yet at runtime a call to `sim.inspect(heap.we)` still hits a `KeyError`.

Gamrix, 2016-08-12:
When did you last update PyRTL?

jemcmahan13, 2016-08-12:
I was on ad4fcc0e4b085cbf0cb5377824e6b2122dfbb1cf at the time I pushed the issue, but the most recent commit doesn't change it.

Yes; it's totally fine without the optimize, and I can keep doing that --- it's just much slower for collecting statistics on non-trivial programs.

Gamrix, 2016-08-12:
No, you need to inspect the probe itself. the wire that is connected to a probe. The wire that is being probed is still a normal wire, and therefore still subject to removal due to optimization.

You will need to give the probe a name to be able to reference it (easily).

Gamrix, 2016-08-17:
Did this work @jemcmahan13 ?

closed 2016-08-31 by Gamrix