UCSBarchlab/PyRTL #223: Write Overflow Detection Wrapper for addition

Gamrix, 2016-08-22:
Overflow detection is a modular addition to addition (pun intended), one that is easy to implement, but also one that new people often get wrong. This is because they assume that the top bit of the result will always signal whether there is an overlflow, which in fact is not the case. It would be great to implement an overflow detection wrapper so that people do not mess up overflow.

Gamrix, 2016-08-22:
I am thinking of some kind of function like the following 
```python
def detect_overflow(adder_res)
   .....
    return adder_res[:-1], overflow

#then called this way
res, overflow = detect_overflow(kogge_stone(a0, a1))
```

nvandervoort, 2016-08-30:
@Gamrix don't you need to be able to access `a0` and `a1` directly in order to detect positive and negative overflow?

Gamrix, 2016-08-30:
You're right

nvandervoort, 2016-08-30:
Is there a way to that as the desired function is written above, or would the have to be passed in directly?

nvandervoort, 2016-08-31:
@Gamrix I wrote one for addition and made a pull request. I wrote one for subtraction too, for good measure but didn't commit it. Let me know if you want that too.

abe-k, 2016-11-01:
Perhaps this would be cleaner as a higher-order function, used something like this:
```
out, overflow = detect_add_overflow(kogge_stone)(in1, in2)
```
especially since this would let it be used to make a permanently-checked function:
```
kogge_stone_checked = detect_add_overflow(kogge_stone)
...
out, overflow = kogge_stone_checked(in1, in2)
```
or even as a decorator:
```
@detect_add_overflow
def my_checked_add(in1, in2):
    ...
```
May I also suggest `signed=false` as default? I feel that hardware is more likely dealing with unsigned values.

Gamrix, 2016-11-01:
Sounds good :)

nvandervoort, 2016-11-01:
@abe-k I have a few days owed to ArchLab, I'll finalize this function and add test iterations as requested by @Gamrix then. Sound good?