UCSBarchlab/PyRTL #228: Still no support for signed addition

jemcmahan13, 2017-06-24:
As discussed in #59, values are treated as unsigned and helper functions are (were) to be provided for signed arithmetic and comparisons. This is true for multipliers, but not for adders. The adders are currently all 'unsigned' in that the uppermost bit is simply the carry-bit from the MSB of the add. For two's compliment, the add should behave as if negative inputs had 1's to the left of their MSB.

Example below shows 8-bit `-1 + 0` resulting in 0xFF, the correct value for an _8_-bit wire, when the answer is 9 bits.

As a note, I think this escaped notice for 2 reasons: 
1) If bitwidth is kept the same, then doing(a+b)[:-1] does, in fact, give the right answer, because the leading zero is dropped; it only produces incorrect results when you keep the extra bit from the add.
2) Leading 0's are not displayed in the trace renderer, so we see `0xFF` instead of `0x0FF`. Without examining bitwidth, the first looks like -1.

```
from pyrtl import *

a = Const(-1, bitwidth=8)
b = Input(8, 'b')
c = Output(name='c')
c <<= a + b

sim_trace = SimulationTrace()
sim = FastSimulation(tracer=sim_trace)

sim.step({b: 0})

sim_trace.render_trace()
print(sim.inspect(c), len(c))  # print the value on wire 'c'
print(Const(-1, bitwidth=len(c)).val)  # print the correct value for -1 on a wire of c's bitwidth
```

timsherwood, 2018-01-09:
Do you have any suggestion on how to fix this up?  I added support for signed and unsigned shift with an explicit function.  We could add a "signed_add(a,b)" which is just "a+b[:-1]"?   We have "signed_lt", "signed_le", etc.  We also have "mult_signed" which I think should be "signed_mult" :)

jemcmahan13, 2018-01-09:
I think signed addition should still add an extra bit. With that, a user can decide how they want to handle overflow. We would need to sign-extend both inputs by one bit, add, then cut the top bit off the result (I think that's the correct behavior).

timsherwood, 2018-01-09:
I think we can swing that

commit 295dccfe (2018-01-10): adds signed add to fix #228, convenience function for negative integer printing
closed 2018-01-10 by timsherwood