UCSBarchlab/PyRTL #230: Error including pyrtl.analysis file

msharmavikram, 2018-01-04:
Hey, 

I am evaluating an high level architecture for my course project and found this library. I  am exploring how to use this library. I followed the instruction as mentioned int the readme and I am stuck when i try importing analysis library. Could you please help what is wrong ? 

```
Python 3.5.2 (default, Nov 23 2017, 16:37:01) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyrtl import * 
>>> from pyrtl.analysis import area_estimation
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'pyrtl.analysis'

```

Gamrix, 2018-01-09:
@timsherwood For some reason, subfolders are not being packaged up in when we release pyrtl

msharmavikram, 2018-01-09:
Yes. pip install does not have required files. I had to manually copy the file pyrtl files to the dist-packages folder. That helped to work. 

Bottleline - pip install pyrtl is a bad way of installing pyrtl! Do a copy paste to dist-packages folder.

timsherwood, 2018-01-09:
That is a bug, sorry about that.  I'll try and fix it now.

closed 2018-01-09 by timsherwood

timsherwood, 2018-01-09:
Okay, I *think* I fixed it but I don't have time at the moment to try it from a clean dist -- let us know if it is not working for you!  Thank you for the bug report :)

Gamrix, 2018-01-09:
I tested, it works. 
`>>> from pyrtl.analysis import estimate`

commit 83cddb47 (2018-01-09): added recursive packages to fix #230