UCSBarchlab/PyRTL #232: Register default values overwritten immediately

abe-k, 2018-01-29:
In `Simulation`, defining a default value for a register (either in `default_value` or `register_value_map`) does not do what I would expect. Because register updates are processed before combinational logic, the default values are immediately overwritten with whatever values were in their inputs. Consider a simple example:
```
a = Input(3)
b = Register(3)
c = Output(3, name='o')
b.next <<= a
c <<= b
sim = Simulation(register_value_map={b: 5})
sim.step({a: 1})
sim.step({a: 2})
print(sim.tracer.trace['o'])
```
I would expect the output of this to be `[5, 1]`, but it is instead `[0, 1]` because the register is updated with the default value of 0 from the input before its default value of 5 has a chance to propagate to the output.

`FastSimulation` handles register updates differently and so behaves in the manner I'd expect. `CompiledSimulation` currently follows the `Simulation` semantics.

In issue #151, memories were changed to update after the combinational logic, but registers were left at the beginning. Do we want to similarly move register updates to the end, or should `FastSimulation` be changed to match the other simulators? What are Verilog's semantics for an `initial` block?

abe-k, 2018-01-29:
I've put some more thought into this, and I'm pretty sure that the behavior of `Simulation` is wrong, and that the behavior of `FastSimulation` is one of two valid options. Consider a block with the following structure (not intended to be runnable PyRTL):
```
inp = Input()
out_reg = Output()
out_inp = Output()
reg = Register()
next = combo_logic(reg, inp)
reg.next <<= next
out_reg <<= reg
out_inp <<= inp
```
Running this on hardware would give us a timing diagram as follows: 
[image]
where R is the reset value of the register, X Y Z are the various inputs passed, and A B C are the results of `combo_logic`. I've added a slight delay between the updates to `reg` and `next` to represent the propagation delay of the logic. It doesn't matter exactly when `inp` updates, so I've put it on the negative clock edge. (If I've gotten the behavior of the hardware wrong, please let me know, and please ignore everything following.)

We want to make sure that the values in one step of a trace would actually occur at the same time; that is, we want each step of the trace to correspond to a single vertical line in the timing diagram. `FastSimulation` currently does so correctly: in this example it would output X for `inp` and `out_inp` and R for `reg` and `out_reg` on the first step, Y and A similarly on the second, etc, effectively sampling just before the rising edge of the clock. `Simulation` however, would first output `default_value` (e.g. 0) for `reg` and `out_reg`, which is not a value they necessarily ever assume, along with X for `inp` and `out_inp`. On the next step, since it used `default_value` instead of R in computing `next`, its output will be completely incorrect if R is not `default_value`.

Besides the behavior currently exhibited by `FastSimulation`, there is one other reasonable-seeming one, and that is to sample immediately after the rising edge of the clock, so on the first output we would have `reg` equal to A while `out_reg` was still `R`. This behavior appears to be the one that was intended  in implementing `FastSimulation`, as evidenced by line 422, `self.context.update(self.regs)`. The context happens to be overwritten with the old values of the registers on the next line, but this appears accidental. The downside of the intended behavior is that when looking at the registers, we never see the initial value, though its effects are observable in the output.

I suggest that `Simulation` and `CompiledSimulation` be changed to have the same semantics as `FastSimulation`, and that `FastSimulation` retain its current behavior with the removal of `self.context.update(self.regs)`. Changing them in this way only causes two tests to fail, `test_adder_simulation` and `example2-counter.py`, both of which assume that the register value accessible to `inspect` (or `.value[]`) is the same as the one most recently output on the trace. I suggest that these tests be changed.

timsherwood, 2018-03-08:
I certainly agree that the top case should be [5, 1] not [0, 1].  I am having a bit of trouble seeing between options 1 and 2 though.    To me, the "sample immediately after the rising edge of the clock" is where my head is with this.  Is there a simple executable example that has different outputs for option 1 and 2 that you can share?  

Also (unrelated), we pushed a change to the behavior of "None" passed as a tracer to simulation today because there was not really a way to specify that you don't want a tracer.  You should see it in the development branch.

Also, if you had these as branches we could try them out on some of our bigger designs and see if they break?

abe-k, 2018-03-08:
The difference between the two options I presented is only visible when looking directly at the registers. Consider the following design:
```
from pyrtl import *

inp = Input(3, 'inp')
reg = Register(3, 'reg')
out1 = Output(3, 'out1')
out2 = Output(3, 'out2')

reg.next <<= inp
out1 <<= inp
out2 <<= reg

sim = FastSimulation(register_value_map={reg: 7})
sim.step({inp: 1})
print(sim.inspect(reg))
sim.step({inp: 2})
print(sim.inspect(reg))
sim.tracer.print_trace(compact=True)
```
Currently this prints `7`, `1`, and then the trace
```
 inp 12
out1 12
out2 71
 reg 71
```
If instead, we were sampling immediately after the clock edge, we'd see the updated register values, so this would print `1`, `2`, and then the trace
```
 inp 12
out1 12
out2 71
 reg 12
```
Note that because the sample is *immediately* after the clock edge, the new register value has not propagated to `out2`. Therefore, I prefer the first option (current FastSim behavior), as it represents a longer-lasting state.

Since CompiledSimulation cannot look directly at the registers, it would be the same under either approach.

I'll be sure to add None-tracer support to CompiledSimulation.

I'll do a pull request for the simulation updates soon—I was hoping to get consensus on this timing issue first.

timsherwood, 2018-03-08:
I agree -- the 7, 1 case should be the right answer.  Thank you for that example!  Actually, it might be helpful for you to write a paragraph describing the above idea of "when" a step coresponds to for the more hardware oriented people in either the conversion to verilog example or in the simulation doc string.  If I can get it wrong, I am guessing it is not obvious to others :)

I am surprised that this would have any impact on the examples or existing tests (well maybe less so the tests).  I need to look at that.

abe-k, 2018-03-08:
I've updated Simulation to match the behavior of FastSimulation, and now all three simulators are consistent on the example I gave. Conveniently, this also fixes the existing test and example that my earlier timing change broke.

I'd be glad to write up some documentation of the timing, but I'm not sure what sort of explanation would be helpful. Are you just looking for something like "The register values after each simulation step are from before the register has latched in its new values," or something more detailed?

mdko, 2021-09-25:
Added documentation in 8b42f566a3c2c23de21f1b534900232219a3b313, so I think we can close this.

closed 2021-09-25 by mdko