UCSBarchlab/PyRTL #237: Tests need to fail gracefully if external dependencies are not installed

timsherwood, 2018-05-31:
The calls to gcc broke my ability to run tests on my mac (which is important).   We need to come up with a good solution to this please :)

======================================================================
ERROR: test_multiply_simulation (tests.test_compilesim.TestTraceWithBasicOpsCompiledSimulation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/sherwood/Desktop/work-stuff/GitProjects/PyRTL/tests/test_compilesim.py", line 71, in test_multiply_simulation
    self.check_trace('o 01377777\n')
  File "/Users/sherwood/Desktop/work-stuff/GitProjects/PyRTL/tests/test_compilesim.py", line 19, in check_trace
    sim = self.sim(tracer=sim_trace)
  File "/Users/sherwood/Desktop/work-stuff/GitProjects/PyRTL/pyrtl/compilesim.py", line 104, in __init__
    self._create_dll()
  File "/Users/sherwood/Desktop/work-stuff/GitProjects/PyRTL/pyrtl/compilesim.py", line 223, in _create_dll
    ], shell=(platform.system() == 'Windows'))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/subprocess.py", line 558, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gcc', '-O0', '-march=native', '-std=c99', '-m64', '-shared', '-fPIC', '-mcmodel=medium', '/var/folders/v3/yfzy2lcs33n5r0ngkq_q5hgw0000gp/T/tmpow2pj0p5/pyrtlsim.c', '-o', '/var/folders/v3/yfzy2lcs33n5r0ngkq_q5hgw0000gp/T/tmpow2pj0p5/pyrtlsim.so']' returned non-zero exit status 1

abe-k, 2018-05-31:
When called on a system without gcc, CompiledSimulation should throw an error. If we don't want these tests to fail, I think the best choice is just to skip them. We can skip the entire `test_compilesim.py` file by adding at the beginning:
```
import subprocess
try:
    subprocess.check_output(['gcc', '--version'])
except OSError:
    raise unittest.SkipTest('CompiledSimulation testing requires gcc')
```

timsherwood, 2018-06-02:
Actually there is "gcc" install on apple but it is a gcc front end to clang (for some reason).  The error gcc is throwing is below.  I have added code that will check if "LLVM" is the in the version string of gcc and if it is it will not test CompiledSim.  Obviously this is all a bit hacky, but for now I think it is fine.

/tmpm89ld7g9/pyrtlsim.c
ld: illegal text-relocation to '_sim_run_step.w0_r' in /var/folders/v3/yfzy2lcs33n5r0ngkq_q5hgw0000gp/T/pyrtlsim-1624e5.o from '_sim_run_step' in /var/folders/v3/yfzy2lcs33n5r0ngkq_q5hgw0000gp/T/pyrtlsim-1624e5.o for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

closed 2018-06-02 by timsherwood

abe-k, 2018-06-02:
Using clang works for me (clang 6.0.1 on Ubuntu). I couldn't find much information on that error message, but I saw a claim that it was due to the Mach-O format not supporting that relocation, which would explain it being a macOS-only issue.

commit 263cea35 (2018-06-02): Conditionally calls test of CompiledSim only for capable machines (fixes #237)