UCSBarchlab/PyRTL #249: Addressing RomBlock at empty address throws exception

jemcmahan13, 2018-11-14:
If you index into a RomBlock at a "valid" address that is beyond the data with which it was initialized, an exception is thrown. To have the same interface as MemBlocks, this read should be allowed (and should presumably return 0).

Below, address `3` is within the address space of 256, but throws an exception because there is no data there.
```
from pyrtl import *

rb = RomBlock(8, 8, [1,2,3])
out = Output(8, 'out')
out <<= rb[Const(3)]

sim = FastSimulation()
sim.step({})
```
`pyrtl.pyrtlexceptions.PyrtlError: RomBlock index is invalid`

jemcmahan13, 2018-11-14:
Additional problem: running `output_to_verilog` with a RomBlock actually throws this error if the data doesn't fill the RomBlock.

abe-k, 2018-11-15:
It appears that this RomBlock behavior is intentional, as `test_simulation.py` has `test_rom_out_of_range_error` which expects the error to be thrown.

I would agree with the suggestion that we change the intended behavior to default-zero rather than default-error. This would also make CompiledSimulation match the others, as it already has default-zero behavior for the sake of efficiency.

jemcmahan13, 2018-11-17:
If you keep it, `output_to_verilog` needs to be fixed. It tries to read past the end of the initialized data and throws an error.

commit 30650679 (2019-05-12): added pad_with_zeros option to rom, fixes #249
closed 2019-05-12 by timsherwood