UCSBarchlab/PyRTL #256: Using matrix of wirevectors resulted in op_param out of bounds error

JacquelineMai, 2019-02-15:
* Hit an op_param out of bounds error at sim step for matrix multiplication
* Error was resolved when the bitwidth of the input integers was decreased from 13 to 11- output bitwidth was set to 2*(input width)+1 in each case.
* Error can be replicated with smallCopy.py in Jacqui branch of PyRTL-research/turtle
   * Must have access to the PyRTL-research repo

abe-k, 2019-02-17:
The problem appears to be the use of `value=pyrtl.wire.Const(0)` in the parameter list for `PyRTLMatrix.__init__`. Because default parameter values are only evaluated when the function is defined, only a single `Const` instance is created for all matrices, so changing `value.bitwidth` affects all the matrices. The specific error is raised because a slice being performed in one matrix fails due to another matrix having reduced the bitwidth below what the slice required. The issue can be resolved by having a sentinel value (such as `None`) as the default value, and checking in the function body for this sentinel, then creating a new `Const` instance, which will not be shared with any other matrix. (By the way, using mutable default arguments is [the most common "gotcha" in Python](https://docs.python-guide.org/writing/gotchas/).)

JacquelineMai, 2019-03-03:
This was the problem! Thank you

closed 2019-03-03 by JacquelineMai