UCSBarchlab/PyRTL #258: Error loading pickled working block

JacquelineMai, 2019-03-03:
On branch Jacqui of PyRTL-research/turtle, file smallPickle.py:

* Pickled working block of smallCopy, trying to load it in another file
* First used `pyrtl.set_working_block(pickle.load(fileObject))`, but function expected a Block as an argument
* Used `Block(pickle.load(fileObject))` to cast loaded object to a Block, but now getting error `__init__() takes 1 positional argument but 2 were given`. The object being loaded from the pickle has type `class 'function'`. This is not the right type, but it only looks like one argument.
* Not sure how to interpret these errors or proceed with loading block

abe-k, 2019-03-03:
Looks like this is just a typo: in smallCopy.py, you have `pickle.dump(pyrtl.working_block, fileObject)` instead of `pickle.dump(pyrtl.working_block(), fileObject)`, so you're pickling the function `working_block` rather than the block it returns.

JacquelineMai, 2019-03-04:
Ah yes, this was the problem. Thanks again

closed 2019-03-04 by JacquelineMai