UCSBarchlab/PyRTL #260: Use of Memory results in TypeError

DKupshM, 2019-04-09:
Using Memory giving Error

```
Traceback (most recent call last):
  
File "/Users/dylankupsh/Desktop/CS/cs190N/PyRTL-research/turtle/Group/PyRTLNeuralNetwork.py", line 163, in <module>
    'input', imageToArray(inputs[0])[:size])
 
 File "/Users/dylankupsh/anaconda3/lib/python3.6/site-packages/pyrtl/simulation.py", line 195, in step
    self._execute(net)
  
File "/Users/dylankupsh/anaconda3/lib/python3.6/site-packages/pyrtl/simulation.py", line 276, in _execute
    self.value[net.dests[0]] = self._sanitize(result, net.dests[0])
  
File "/Users/dylankupsh/anaconda3/lib/python3.6/site-packages/pyrtl/simulation.py", line 241, in _sanitize
    return val & wirevector.bitmask

TypeError: ufunc 'bitwise_and' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

```
When printing val
114133

When printing wirevector.bitmask
18446744073709551615

This happens in simulation.step() portion of the code. When the size of the input was lowered then it ended up working.

The code also passes some test cases that we created for the individual storing of matrices.

Error can be replicated within PyRTLNeuralNetwork.py in Dylan branch of PyRTL-research/turtle
- Must have access to the PyRTL-research repo

abe-k, 2019-04-09:
It looks like this is a bug in NumPy: https://github.com/numpy/numpy/issues/2955

To avoid it, be sure to cast the values passed into the simulation (in this case, via `memory_value_map`) to `int` rather than leaving them as NumPy types.

(As a side note, when raising an issue, _please_ try to provide a minimal reproducible example. The program you gave took a very long time to run and used more memory than my laptop had available, so I almost didn't bother running it.)

DKupshM, 2019-04-20:
Thanks for everything! It works perfectly after the fix. Will provide a minimum example next time

closed 2019-04-20 by DKupshM