UCSBarchlab/PyRTL #317: Allow Use of Negative IntEnums

mdko, 2021-02-24:
When using an IntEnum member in a location where it will be implicitly turned into a `Const` and call `infer_val_and_bitwidth`, members whose values are negative will throw an error in PyRTL because there's no way to specify that this Const instantiation should have `signed=True`. One solution is to have a special check for whether the value is an IntEnum, and if so, set the bitwidth to be `max(YourEnum).bit_length()`.

For example, the following code

```python
from enum import IntEnum
import pyrtl

class Causes(IntEnum):
    BadAddress = 0
    IllegalInstruction = -1
    Timeout = 3

if __name__ == "__main__":
    i = pyrtl.Input(max(Causes).bit_length(), 'i')
    o = pyrtl.Output(1, 'o')
    with pyrtl.conditional_assignment:
        with (i == Causes.BadAddress) | (i == Causes.IllegalInstruction):
            o |= 0
        with pyrtl.otherwise:
            o |= 1

    sim = pyrtl.Simulation()
    sim.step_multiple({
        'i': [v for v in Causes]
    })
    sim.tracer.render_trace()
```

produces
```bash
❯ python3 ex1.py 
Traceback (most recent call last):
  File "ex1.py", line 13, in <module>
    with (i == Causes.BadAddress) | (i == Causes.IllegalInstruction):
  File "/Users/michael/PyRTL/pyrtl/wire.py", line 332, in __eq__
    return self._two_var_op(other, '=')
  File "/Users/michael/PyRTL/pyrtl/wire.py", line 195, in _two_var_op
    a, b = self, as_wires(other)
  File "/Users/michael/PyRTL/pyrtl/corecircuits.py", line 388, in as_wires
    return Const(val, bitwidth=bitwidth, block=block)
  File "/Users/michael/PyRTL/pyrtl/wire.py", line 603, in __init__
    num, bitwidth = infer_val_and_bitwidth(val, bitwidth, signed)
  File "/Users/michael/PyRTL/pyrtl/helperfuncs.py", line 469, in infer_val_and_bitwidth
    return _convert_int(rawinput, bitwidth, signed)
  File "/Users/michael/PyRTL/pyrtl/helperfuncs.py", line 503, in _convert_int
    raise PyrtlError('negative constants require either signed=True or specified bitwidth')
pyrtl.pyrtlexceptions.PyrtlError: negative constants require either signed=True or specified bitwidth
```

timsherwood, 2021-03-06:
Hmm -- I am thinking about this and I think causes a bunch of downstream problems.  Everything in a PyRTL is unsigned for a reason, because when you start mixing zero extend and sign extend in different ways it gets really weird.  There really are no negative numbers, only convenience functions that make dealing with twos complement easier.

In this case the size of your enums are going to be different.  That is fine when everything is a non-negative number, zero extend "does the right thing" and all your comparisons make sense.  But if you zero extend a negative number it is definitely not the the same number when you are done.  The == operator will zero extend to get them to be the same length, so if `i` is a 32-bit number and `-1` is a 2-bit signed number they when you compare them with == you are going to have a bad time.

Honestly I think in this case I would suggest instead of -1 you use the an integer value which is -1 converted to the bitwidth you will use it with.  There is a pyrtl function `formatted_str_to_val()` that will do that.  For example, if they are three bit constants you are trying to define you can use `formatted_str_to_val('-1', 's3')` which would return 0b111 (better known as 7)!

mdko, 2021-03-06:
Okay, yes I see the issue. Being able to compare a `-1` from the Enum, even it was determined that it had to be, say, 4 bits wide to be totally unique with respect to the other positive members of that same Enum, causes problems during comparison against other arbitrary numbers because, e.g. a 4-bit `-1` is the same as a 15, while an 8-bit `-1` is the same as a 255.

I agree it's probably non-sensical to supply a negative IntEnum, its just surprising where the error is being reported (it seems to imply the user should somehow supply the bitwidth for the corresponding negative IntEnum, where there isn't a way). I'm wondering if a better error message or better documentation might help here instead.

closed 2021-03-06 by mdko