UCSBarchlab/PyRTL #324: Don't Cares in Pattern Matching

BracketMaster, 2021-03-04:
I can't seem to fine don't care pattern matching capability in PyRTL
like there is in Chisel.

for example
```
def ADD       = BitPat("b011111???????????????0100001010?")
```

Does PyRTL currently have support for such a pattern?

https://github.com/antonblanchard/chiselwatt/blob/master/src/main/scala/Instructions.scala#L5

timsherwood, 2021-03-04:
Hi!  That is a cool, I had not seen that.  That is not in PyRTL right now but I think we could do it pretty quickly...

timsherwood, 2021-03-04:
I think we can add this to the mainline pyrtl soon, but in the meantime you can use the function below (it probably needs more error handling and tests before I can add it to the proper pyrtl library).   If this does not address the use case you are imagining let me know.   Thanks!

`import pyrtl

def match_pattern(w, bitpattern):
    ''' example: match_pattern(thing_to_test, '0101??11') '''
    assert len(w) == len(bitpattern)
    # make a list of the bits of w that are supposed to be '0' and supposed to be '1'
    # remember to walk through the pattern backwards (since the lsb is on the right!)
    zero_bits = [w[index] for index,x in enumerate(bitpattern[::-1]) if x == '0']
    one_bits = [w[index] for index,x in enumerate(bitpattern[::-1]) if x == '1']
    # check that all the one bits are one, and all the zero bits are zero (the '*' here
    # just unpacks the list into the arguments of the function)
    return pyrtl.rtl_all(*one_bits) & ~pyrtl.rtl_any(*zero_bits)

x = pyrtl.Input(8,'x')
y = pyrtl.Output(1,'y')

y <<= match_pattern(x, '0101??11')
sim = pyrtl.Simulation()

sim.step_multiple({'x': [ 0b01011111, 0b01010111, 0b01010011, 0b00000000, 0b00010111, 0b01010010]})
sim.tracer.render_trace()
`

BracketMaster, 2021-03-04:
that will probably work.
also - slightly off-topic - I had seen JIT to C mentioned [here](http://ucsbarchlab.github.io/PyRTL/)
but nowhere else? Is JIT to C a feature?

timsherwood, 2021-03-04:
Oh weird, I didn't realize that is not showing up in the documentation (I just checked and don't see it).  Instead of creating a `Simulation` class to simulate the design you can use `FastSimulation` which will JIT to python or `CompiledSimulation` which will JIT to C (each has a longer time to compile and start but runs faster per instruction.  BTW we are overdue for an update to pipy so if you hit a problem don't be afraid to pull from development -- we keep it very stable.  

Thanks for pointing it out the missing documentation.  Not sure why that is not showing up in the docs, I will add that as a bug.   Unrelated, let us know if you build anything cool because we always like hearing about interesting projects :)

closed 2021-03-05 by timsherwood

timsherwood, 2021-03-06:
BTW -- I was so long overdue for an update to the PYPI I just added in the match_bitpattern into pyrtl and pushed everything out to pypi.  If you pip install (or upgrade) pyrtl it should now be version 0.9 and you will find match_bitpattern in there!

commit 15b8ae1e (2021-03-05): adds tests for match_bitpattern, fixes #324