UCSBarchlab/PyRTL #421: Python 3.10 Support

benjaminmordaunt, 2022-07-26:
When trying to run the basic FIR example, Python 3.10 complains:
```
File /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pyrtl/simulation.py:983, in <module>
    979             return s
    980     return [tryint(c) for c in re.split('([0-9]+)', w)]
--> 983 class TraceStorage(collections.Mapping):
    984     __slots__ = ('__data',)
    986     def __init__(self, wvs):
```

It seems that accessing `collections.Mapping` in this was was previously deprecated and is now removed.

timsherwood, 2022-07-29:
Hi @benjaminmordaunt, thank you for the clear report -- in fact this issue should already be fixed in the development branch.  If you could try that out I am hoping it addresses your problem ... it appeared to pass all test in 3.10 here just now (including the fir example) although I don't have those test ready for release yet.  I was planing to do a major release around the end of summer which would solve this problem on the PyPI release.  Let me know if that helps!  I will leave this issue open until that major release in case anyone else hits this problem.

All the best,

-Tim

fdxmw, 2024-05-31:
PyRTL 0.10.2 and newer should support Python 3.10. Please reopen if there are further issues, thanks!

closed 2024-05-31 by fdxmw