UCSBarchlab/PyRTL #457: Hello from the OSPO!

LauraLangdon, 2024-12-11:
Hello! I’m the Community Manager for the UC’s new [Open Source Program Office (OSPO) Network](https://ucospo.net/), and I’m reaching out to folks who have UC-affiliated public repos on GitHub to introduce myself and see if the OSPO network can support your open source efforts. I’ll be helping maintainers who would like to increase the number of contributors to their projects, and I wondered if you might be interested in that kind of endeavor for this project. Absolutely no pressure, just a service available to you if you’d find it useful! 😊
 
Let me know if you'd like to connect further, and I'd be happy to discuss how we might be able to support your open source work.
 
Cheers,
Laura

fdxmw, 2025-01-03:
Hi Laura, we are definitely interested in having more contributors to PyRTL. I'd love to hear about your services, maybe we can find some time for a video chat?

LauraLangdon, 2025-01-08:
Awesome! My email address is lalangdon@ucdavis.edu if you'd like to send me some dates and times you'd be available for a video chat, or if it's easier you can [grab time on my calendar](https://book.vimcal.com/p/lauralangdon/30-minute-meeting-0af0d). 😊